DeqingSun/ch55xduino #1: correct flash size

nerdralph, 2020-06-30:
The total flash size on the CH552 and CH554 may be 16kB, but 2kB is used by the bootloader, leaving 14kB.  Details are in the ch554_sdcc issue.
https://github.com/Blinkinlabs/ch554_sdcc/issues/16

DeqingSun, 2020-06-30:
You are right, On chapter 2 of datasheet, 16K can be configured to all app space or 14K app + 2K bootloader. I supposed all ch55xduino uses a 2K bootloader.
On Chapter 6.2, 
```
Code Flash includes the application code of the low address area and the boot code of the high address area, and it is also possible to combine two block areas to save a single application code.
For CH551, Code Flash's application code area is only 10KB.
```
Seems CH551 doesn't count 2K bootloader into the 10K specs.

DeqingSun, 2020-06-30:
https://github.com/DeqingSun/ch55xduino/commit/cf9a28f32a92046b09fed70ed11242a01bc550eb
should fix this.

closed 2020-06-30 by DeqingSun