DeqingSun/ch55xduino #3: optimize digitalWrite

nerdralph, 2020-06-30:
As you have noticed, lookup tables in flash for pin mapping contributes to code bloat.  I suggest macros like Wiring uses instead of the lookup tables in flash.
https://github.com/WiringProject/Wiring/blob/master/framework/hardware/Atmel/ATmegaXX8P-PA/BoardDefs.h
I think you should be able to get digitalWrite to compile to a single setb or clr instruction.

DeqingSun, 2020-06-30:
I checked the map file, the lookup tables take 160 bytes, no matter if we use digitalWrite or not. When wiring_digital is linked, an additional of 395 bytes will be used. 

I think it makes sense to use macros to replace lookup tables, no code space will be used when wiring_digital is not linked. Instead, there can be a ``wiring_digital_fast`` file that uses a lot of macros to reduce code space.

DeqingSun, 2020-07-04:
I tried to use Marco to do so. However, the code gets significantly bigger.
```
#define digitalPinToPort(P) ( ((P%10)<8) ? (P0PORT+(P/10)) : NOT_A_PORT )
#define digitalPinToBitMask(P) ( ((P%10)<8) ? (1<<(P%10)) : 0 )
```
Seems we need to find a way to make SDCC not link the array if not needed.

DeqingSun, 2020-07-04:
fixed

https://github.com/DeqingSun/ch55xduino/commit/f8a01ef8f27b5d900706a101bfc3b8386731f106

If you don't use any wire analog or digital functions. The LUT won't be linked.

closed 2020-07-04 by DeqingSun