DeqingSun/ch55xduino #4: add option to disable CDC serial

nerdralph, 2020-06-30:
The Arduino Blink sketch compiles to 5333 bytes of flash, and about 4kB of that seems to be the CDC serial code.  Using ch554_sdcc, examples/blink compiles to 700 bytes.  By leaving out the UART code in debug.c, I reduced it to 232 bytes.

I realize that without the CDC serial code, a manual power cycle into boot mode would be required before uploading new code.  I'm hoping there is a way to leave the device in DFU, while still running the user code.  Some investigation into how the bootloader works would be required.

DeqingSun, 2020-06-30:
In the next version, there will be an option to leave all USB work to the user. So the user will do the USB handler within the sketch folder, with the freedom to make whatever USB device they like, without modifying the board support file.

There is some work on decoding the bootloader code on https://www.mikrocontroller.net/topic/462538 . However, the bootloader is subject to change. The user can write his own bootloader jump code that fits his need, with an empty USB handler.

DeqingSun, 2020-07-03:
> The Arduino Blink sketch compiles to 5333 bytes of flash, and about 4kB of that seems to be the CDC serial code. Using ch554_sdcc, examples/blink compiles to 700 bytes. By leaving out the UART code [...]

In my current test an empty sketch compiled for 2428 bytes. about 200 bytes are lookup table, and serial0 takes about 600 bytes.

DeqingSun, 2020-07-04:
fixed 
https://github.com/DeqingSun/ch55xduino/commit/de114480dc5fa533be5d3353ad079149ab5b48a5

closed 2020-07-04 by DeqingSun