DeqingSun/ch55xduino #8: link error with USER_CODE option

nerdralph, 2020-07-08:
?ASlink-Warning-Undefined Global '_USBInterrupt' referenced by module 'main'

DeqingSun, 2020-07-08:
I didn't find a way for sdcc to do weak declaration. So USBInterrupt is refered in main for user to declare it. 

If you check the 2 examples, they both have it declared. If you don't need USB, just do an empty function.

DeqingSun, 2020-07-08:
maybe it is possible to do overloading by using a library

https://sourceforge.net/p/sdcc/mailman/message/35824985/

DeqingSun, 2020-07-08:
Fixed with https://github.com/DeqingSun/ch55xduino/commit/f3a5db8d16c8a0e36b8c7d2d0726ad286d349db9

An additional weakOverride.c works.

closed 2020-07-08 by DeqingSun

nerdralph, 2020-07-09:
> Fixed with f3a5db8
>
> An additional weakOverride.c works.

Nice.  I've done something similar with my picoUART library.

nerdralph, 2020-07-16:
With an empty USBInterrupt function, it links and uploads, but the code doesn't run; it seems to stay in the bootloader.  After a power cycle, the same thing.

DeqingSun, 2020-07-16:
Are you using a delay?

That was a bug because I didn't enable EA by default. Now it should work with https://github.com/DeqingSun/ch55xduino/commit/cde2fd71a72cd5b44f061124bd7d433ee1aefbf6

nerdralph, 2020-07-16:
> @DeqingSun <https://github.com/DeqingSun> Are you using a delay?
> with cde2fd7

Yes, I was testing with a basic blink sketch.