DeqingSun/ch55xduino #12: No support for Linux x64

zoobab, 2020-08-02:
I tried to install ch55xduino on Arduino IDE under a Linux 64bits, I got the following error:

"Tool MCS51Tools is not available for your operating system"

Any chance to add some Linux support?

DeqingSun, 2020-08-02:
Sure, would you try the sdcc in http://sdcc.sourceforge.net/snap.php
and see which binary worked for you? And also what is your OS version?
Then I'll try to put that tool in this repo.

NoNamedCat, 2020-10-14:
Please, please! I want to program this IC without installing Windous in my PC.
Can you put this tool for linux?

NoNamedCat, 2020-10-14:
I have a 64 bit operating system. I tried the 32 bit version of SDCC and it runs correctly. That version should be more adequate because it runs in 64bits linux too.

DeqingSun, 2020-10-14:
> I have a 64 bit operating system. I tried the 32 bit version of SDCC and it runs correctly. That version should be more adequate because it runs in 64bits linux too.

@NoNamedCat Would you share the SDCC link you used?

NoNamedCat, 2020-10-14:
Sure! http://sourceforge.net/projects/sdcc/files/snapshot_builds/i386-unknown-linux2.5/sdcc-snapshot-i386-unknown-linux2.5-20201014-11918.tar.bz2/download

DeqingSun, 2020-10-19:
@zoobab @NoNamedCat Now the linux support is added. And I've tested on Ubuntu. 
Let me know if it works for you.

NoNamedCat, 2020-10-20:
Only one thing:
 MCS51Tools in the JSON file for Boards manager dont have suport yet for Liunux. Thanks a lot!

DeqingSun, 2020-10-20:
https://github.com/DeqingSun/ch55xduino/blob/8ba72cbb4e5d0c70b0dd8debe9566a9644cd524b/package_ch55xduino_mcs51_index.json#L160

Maybe you delete all old cache and try again?

NoNamedCat, 2020-10-20:
I already cleared the Arduino cache and still can't install ch55xduino in the development environment. I think you should add those lines a little lower in the JSON file:

`                {
                    "name": "MCS51Tools",
                    "version": "2020.07.07",
                    "systems": [
                        {
                            "host": "x86_64-apple-darwin",
                            "url": "https://github.com/DeqingSun/ch55xduino/releases/download/0.0.4/ch55xduino-tools_macosx-2020.07.07.tar.bz2",
                            "archiveFileName": "ch55xduino-tools_macosx-2020.07.07.tar.bz2",
                            "checksum": "SHA-256:b3b14dbb9b6258530f5e70822c04d25210bd0a843902a8dbbfa35314aafd4aa6",
                            "size": "16595"
                        },{
                            "host": "i686-mingw32",
                            "url": "https://github.com/DeqingSun/ch55xduino/releases/download/0.0.4/ch55xduino-tools_mingw32-2020.07.07.tar.bz2",
                            "archiveFileName": "ch55xduino-tools_mingw32-2020.07.07.tar.bz2",
                            "checksum": "SHA-256:5745507f17ff380113496cf40b3b89bde0d4b0f45a4b58ad15fdbd41f9c4c4ca",
                            "size": "665584"
                        }
                    ]
                },
                {
                    "name": "MCS51Tools",
                    "version": "2020.05.31",
                    "systems": [
                        {
                            "host": "x86_64-apple-darwin",
                            "url": "https://github.com/DeqingSun/ch55xduino/releases/download/0.0.1/ch55xduino-tools_macosx-2020.05.31.tar.bz2",
                            "archiveFileName": "ch55xduino-tools_macosx-2020.05.31.tar.bz2",
                            "checksum": "SHA-256:7516a09d14cac399f80ce8b1c098ce24835758d5e924b9bf0b3fb4958add252a",
                            "size": "10894"
                        },{
                            "host": "i686-mingw32",
                            "url": "https://github.com/DeqingSun/ch55xduino/releases/download/0.0.1/ch55xduino-tools_mingw32-2020.05.31.tar.bz2",
                            "archiveFileName": "ch55xduino-tools_mingw32-2020.05.31.tar.bz2",
                            "checksum": "SHA-256:bd2a98d92888b021adc68c05a0d7a995bb1814bfb076d3805d27b9ffd65bfbcb",
                            "size": "506274"
                        }
                    ]
                },`

NoNamedCat, 2020-10-20:
Here:

https://github.com/DeqingSun/ch55xduino/blob/8ba72cbb4e5d0c70b0dd8debe9566a9644cd524b/package_ch55xduino_mcs51_index.json#L173

https://github.com/DeqingSun/ch55xduino/blob/8ba72cbb4e5d0c70b0dd8debe9566a9644cd524b/package_ch55xduino_mcs51_index.json#L192

NoNamedCat, 2020-10-20:
Ok! I have now intalled the ch55xduino!
Man!!! Only gratitude for all this work. TKS a lot!!

DeqingSun, 2020-10-20:
@NoNamedCat Great you installed it.

For anyone facing a similar problem. The linux will work for 0.0.6 or newer release, not older ones.

closed 2020-10-20 by DeqingSun