DeqingSun/ch55xduino #13: CH552 CMISIS DAP

sabas1080, 2020-08-29:
Hi @DeqingSun 

First of all thank you for your work

I am triying your code for CMISIS DAP with CH552, but but I can't see the usb port of the device, is there any reference that it should appear?

Thanks

DeqingSun, 2020-08-29:
What you mean usb port? It should appear as an HID device.

sabas1080, 2020-08-29:
@DeqingSun I have bootloadwer 2.4, it can be that?

when the chip is in bootloader mode it appears the device USB and I can program with wchisptool, but when I load the compiled example, everything disappears

DeqingSun, 2020-08-29:
It seems you are using a newer chip. I'll check if https://github.com/ishiyakazuo/ch55x-isptool/commit/151c55e308c1b6c0e93204069938bc114eda363c works.

sabas1080, 2020-08-29:
@DeqingSun work for me, but i can't get a blink

[image]

My schematic work at 3.3v

[image]

DeqingSun, 2020-08-29:
I just compiled the 2 versions, would you also try if my file worked for you?

Try to use 5V for main chip to see if the USB work properly. CH552 may not work at 24M with 3.3V power supply.

[vnproch55x.exe.zip](https://github.com/DeqingSun/ch55xduino/files/5145898/vnproch55x.exe.zip)
[vnproch55x.zip](https://github.com/DeqingSun/ch55xduino/files/5145899/vnproch55x.zip)

sabas1080, 2020-08-29:
Mac work for me, but Windows no

now I have a blink with 12Mhz

sabas1080, 2020-08-29:
CMSIS DAP working for 12Mhz
[image]

DeqingSun, 2020-08-29:
Good, you may use 16M, that will be a bit faster.

closed 2020-08-29 by sabas1080