DeqingSun/ch55xduino #15: CMSIS DAP drivers windows and PyOCD

sabas1080, 2020-09-01:
Hi @DeqingSun 

I am testing CMISIS DAP with pyOCD with `pyocd list` and and i see it

[image]

but  I have a error

`pyocd gdbserver -t nrf52840`

this is in Mac with https://github.com/mbedmicro/pyOCD

Dapjs
https://github.com/ARMmbed/dapjs

En Windows with Keli
[image]

DeqingSun, 2020-09-01:
I haven't fully tested it with different software. But it worked well with the openocd from Arduino package

sabas1080, 2020-09-02:
Thanks @DeqingSun work en openocd

DeqingSun, 2020-10-26:
Try 
https://github.com/DeqingSun/ch55xduino/tree/playground/ch55xduino/ch55x/libraries/Generic_Examples/examples/05.USB/CMSIS_DAP

And this should work. If you get it work, or not. Please let me know.

sabas1080, 2020-10-27:
Hi @DeqingSun 

I have the next error:

```
➜  ~ pyocd list
  #   Probe                      Unique ID
--------------------------------------------
  0   ElectronicCats CMSIS-DAP   CH55x
➜  ~ pyocd erase -t nrf52840 --chip
0000763:CRITICAL:__main__:No ACK received
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/cmsis_dap_probe.py", line 249, in read_dp
    result = self._link.read_reg(reg_id, now=now)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 836, in read_reg
    return read_reg_cb()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 830, in read_reg_cb
    res = transfer.get_result()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 143, in get_result
    self.daplink.flush()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/utility/concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 667, in flush
    self._read_packet()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/utility/concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 915, in _read_packet
    decoded_data = cmd.decode_data(raw_data)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 449, in decode_data
    data = self._decode_transfer_block_data(data)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 418, in _decode_transfer_block_data
    self._check_response(data[3])
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 341, in _check_response
    raise DAPAccessIntf.TransferError("No ACK received")
pyocd.probe.pydapaccess.dap_access_api.DAPAccessIntf.TransferError: No ACK received

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/__main__.py", line 362, in run
    self._COMMANDS[self._args.cmd](self)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/__main__.py", line 556, in do_erase
    with session:
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/core/session.py", line 302, in __enter__
    self.open()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/core/session.py", line 420, in open
    self._board.init()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/board/board.py", line 85, in init
    self.target.init()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/core/coresight_target.py", line 160, in init
    seq.invoke()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/utility/sequencer.py", line 213, in invoke
    resultSequence.invoke()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 303, in _connect
    connector.connect(self._protocol)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 167, in connect
    self._idr = self.read_idr()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 194, in read_idr
    dpidr = self._probe.read_dp(DP_IDR, now=True)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/cmsis_dap_probe.py", line 251, in read_dp
    six.raise_from(self._convert_exception(error), error)
  File "<string>", line 3, in raise_from
pyocd.core.exceptions.TransferError: No ACK received
```

DeqingSun, 2020-12-03:
Not sure why it doesn't work with pyocd. It did work with Keil.

sabas1080, 2020-12-03:
I'm going to try with keil

sabas1080, 2020-12-08:
I was tried in

Windows 10
Keil 5.27.1

and not work for me, any special configuration? thanks

DeqingSun, 2020-12-08:
I don't have any special configuration. It just worked in Win7.

nerdralph, 2020-12-13:
I also am having problems with pyOCD 0.28.3.  pyOCD list shows the adapter, but trying connect to a target always fails.
```
$ pyocd gdbserver
0000664:WARNING:board:Generic 'cortex_m' target type is selected by default; is
this intentional? You will be able to debug most devices, but not program  flash
. To set the target type use the '--target' argument or 'target_override' option
. Use 'pyocd list --targets' to see available targets types.
0000664:INFO:board:Target type is cortex_m
0000666:CRITICAL:__main__:expected DAP_INFO
Traceback (most recent call last):
  File "c:\users\X\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\probe\cmsis_dap_probe.py", line 148, in open
    self._link.open()
  File "c:\users\X\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\utility\concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "c:\users\X\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\probe\pydapaccess\dap_access_cmsis_dap.py", line 592, in open
    self._packet_count = self._protocol.dap_info(self.ID.MAX_PACKET_COUNT)
  File "c:\users\X\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\probe\pydapaccess\cmsis_dap_core.py", line 138, in dap_info
    raise DAPAccessIntf.DeviceError("expected DAP_INFO")
pyocd.probe.pydapaccess.dap_access_api.DAPAccessIntf.DeviceError: expected DAP_INFO

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "c:\users\X\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\__main__.py", line 398, in run
    self._COMMANDS[self._args.cmd](self)
  File "c:\users\X\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\__main__.py", line 742, in do_gdbserver
    with session:
  File "c:\users\X\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\core\session.py", line 343, in __enter__
    self.open()
  File "c:\users\X\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\core\session.py", line 457, in open
    self._probe.open()
  File "c:\users\X\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\probe\shared_probe_proxy.py", line 54, in open
    self._probe.open()
  File "c:\users\X\appdata\local\programs\python\python37-32\lib\site-packages\pyocd\probe\cmsis_dap_probe.py", line 168, in open
    six.raise_from(self._convert_exception(exc), exc)
  File "<string>", line 3, in raise_from
pyocd.core.exceptions.ProbeError: expected DAP_INFO
```
Looking at pyocd/probe/pydapaccess/cmsis_dap_core.py indicates it didn't receive a 0 (Command.DAP_INFO) in the first byte of the response:
```
        if resp[0] != Command.DAP_INFO:
            # Response is to a different command
            raise DAPAccessIntf.DeviceError("expected DAP_INFO")
```

However looking at the CMSIS_DAP example, I haven't found a bug.  I've compared it to the DAPLink source, and although the CH55x DAP example doesn't implement many of the DAP_INFO commands, it copies the command from the request to the response just like the official firmware.
https://github.com/ARMmbed/DAPLink/blob/master/source/daplink/cmsis-dap/DAP.c

nerdralph, 2020-12-13:
I also get the same error under Linux (Ubuntu 18.04).
`pyocd.probe.pydapaccess.dap_access_api.DAPAccessIntf.DeviceError: expected DAP_INFO`

DeqingSun, 2020-12-14:
@nerdralph Did you get the code from the Arduino installed. For some reason it seems the code was not updated in the release. Will fix in the next release. And for this moment, just grab the code from the Github repo and it should fix this issue.

nerdralph, 2020-12-14:
> 
> 
> @nerdralph Did you get the code from the Arduino installer? For some reason it seems the code was not updated in the release package. Will fix in the next release. And for this moment, just grab the code from the Github repo and it should fix this issue.

I only changed the USB PID.  I didn't notice there was a bug fix in DAP.[ch].  I've just pulled the version from master and will re-build and retry pyOCD.

nerdralph, 2020-12-14:
Copying the CMSIS_DAP example files from master did the trick.
```
$ pyocd gdb -t stm32f051
0000650:INFO:board:Target type is stm32f051
0000732:INFO:dap:DP IDR = 0x0bb11477 (v1 MINDP rev0)
0000834:INFO:ap:AHB-AP#0 IDR = 0x04770021 (AHB-AP var2 rev0)
0000887:INFO:rom_table:AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=02
0 part=440)
0000914:INFO:rom_table:[0]<e000e000:SCS-M0+ class=14 designer=43b part=008>
0000924:INFO:rom_table:[1]<e0001000:DWT-M0+ class=14 designer=43b part=00a>
0000936:INFO:rom_table:[2]<e0002000:BPU class=14 designer=43b part=00b>
0000941:INFO:cortex_m:CPU core #0 is Cortex-M0 r0p0
0000955:INFO:dwt:2 hardware watchpoints
0000971:INFO:fpb:4 hardware breakpoints, 0 literal comparators
0001078:INFO:server:Semihost server started on port 4444 (core 0)
0001084:INFO:gdbserver:GDB server started on port 3333 (core 0)
```

sabas1080, 2020-12-14:
ok for me and working

`pyocd gdb -t nrf52840` and `pyocd erase -t nrf52840`

but with the command flash
`pyocd flash -t nrf52840 Blink.hex`

I have the next error

```
pyocd flash -t nrf52840 bast_frontier_bootloader-0.3.2-115-gd732893-dirty-nosd.hex
0001588:ERROR:board:link exception during target disconnect:
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/__main__.py", line 572, in do_flash
    programmer.program(self._args.file,
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/flash/file_programmer.py", line 158, in program
    self._loader.commit()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/flash/loader.py", line 164, in commit
    perf = builder.program(chip_erase=chipErase,
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/flash/builder.py", line 421, in program
    self._build_sectors_and_pages(keep_unwritten)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/flash/builder.py", line 286, in _build_sectors_and_pages
    fill_end_of_page_gap()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/flash/builder.py", line 264, in fill_end_of_page_gap
    old_data = self.flash.target.read_memory_block8(page_data_end, old_data_len)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/core/soc_target.py", line 170, in read_memory_block8
    return self.selected_core.read_memory_block8(addr, size)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/coresight/cortex_m.py", line 451, in read_memory_block8
    data = self.ap.read_memory_block8(addr, size)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/core/memory_interface.py", line 89, in read_memory_block8
    mem = self.read_memory_block32(addr, size // 4)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/utility/concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/coresight/ap.py", line 972, in _read_memory_block32
    resp += self._read_block32_page(addr, n//4)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/coresight/ap.py", line 930, in _read_block32_page
    resp = self.dp.read_ap_multiple(self.address.address + self._reg_offset + MEM_AP_DRW, size)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/coresight/dap.py", line 642, in read_ap_multiple
    result_cb = self.probe.read_ap_multiple(addr, count, now=False)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/cmsis_dap_probe.py", line 309, in read_ap_multiple
    result = self._link.reg_read_repeat(count, ap_reg, dap_index=0, now=now)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 870, in reg_read_repeat
    transfer = self._write(dap_index, num_repeats, request, None)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/utility/concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 1018, in _write
    self._send_packet()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/utility/concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 963, in _send_packet
    self._read_packet()
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/utility/concurrency.py", line 28, in _locking
    return func(self, *args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 918, in _read_packet
    decoded_data = cmd.decode_data(raw_data)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 451, in decode_data
    data = self._decode_transfer_data(data)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 355, in _decode_transfer_data
    raise ValueError('DAP_TRANSFER response error')
ValueError: DAP_TRANSFER response error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/board/board.py", line 98, in uninit
    self.target.disconnect(resume)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/core/soc_target.py", line 134, in disconnect
    core.disconnect(resume)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/coresight/cortex_m.py", line 324, in disconnect
    self.resume()
[... cut ...]
    decoded_data = cmd.decode_data(raw_data)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 451, in decode_data
    data = self._decode_transfer_data(data)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pyocd/probe/pydapaccess/dap_access_cmsis_dap.py", line 355, in _decode_transfer_data
    raise ValueError('DAP_TRANSFER response error')
ValueError: DAP_TRANSFER response error
```

DeqingSun, 2020-12-14:
@sabas1080  I guess the DAP_SWD_TransferBlock didn't handle the situation well. Will do more investigation.

DeqingSun, 2020-12-15:
@sabas1080 seems to be an DAP_PACKET_COUNT thing. The default is 4 while this implementation didn't do a queue. And \yOcd is trying to run 4 commands before get 4 responses. 
Changing DAP_PACKET_COUNT to 1 seems help. Will test again with different environment.

sabas1080, 2020-12-15:
@sabas1080 nice!, Working for me :D

closed 2020-12-15 by DeqingSun