DeqingSun/ch55xduino #16: USB MIDI?

NoNamedCat, 2020-10-22:
So... I've been playing around with the CH551 and I'm trying to create a MIDI device.

I have tried to port (to CH55Xduino) and compile some code that is on github but using SDCC directly. This is the code that I have tried to compile, and i can compile it and upload it to the IC. But the device is not detected by the computer:

Maybe a little help here?

***********************************************************
***********************************************************
***********************************************************
***********************************************************

__xdata __at (0x0000) uint8_t  Ep0Buffer[DEFAULT_ENDP0_SIZE];    // Endpoint 0 OUT & IN buffer, must be even address
__xdata __at (0x0040) uint8_t  Ep1Buffer[DEFAULT_ENDP1_SIZE];    // Endpoint 1 upload buffer
__xdata __at (0x0080) uint8_t  Ep2Buffer[2*MAX_PACKET_SIZE];  //Endpoint 2 IN & OUT buffer, must be an even address

uint16_t SetupLen;
uint8_t   SetupReq,Count,UsbConfig;
const uint8_t *  pDescr;                             //USB configuration flag
USB_SETUP_REQ   SetupReqBuf;                           //Temporary Setup package
#define UsbSetupBuf  ((PUSB_SETUP_REQ)Ep0Buffer)

/* Device descriptor */
__code uint8_t DevDesc[] = {  
        0x12,0x01,0x10,0x01,
        0x00,0x00,0x00,DEFAULT_ENDP0_SIZE, /* Define it in interface level */
        0x86,0x1a,0x22,0x57,0x00,0x01,0x01,0x02,0x03, /* VID PID bString */
        0x01
         };
        
__code uint8_t CfgDesc[] ={
  0x09,0x02,sizeof(CfgDesc) & 0xff,sizeof(CfgDesc) >> 8,
  0x02,0x01,0x00,0x80,0x32,      //配置描述符（两个接口）
  //以下为接口0（音频接口）描述符
  0x09,0x04,0x00,0x00,0x00,0x01,0x01,0x00,0x00, // USB Audio Class description, no endpoint
  //以下为功能描述符
  0x09,0x24,0x01,0x00,0x01,0x09,0x00,0x01,0x01, //功能描述符，长度大端
  //以下为接口1（MIDI接口描述符）
  0x09,0x04,0x01,0x00,0x02,0x01,0x03,0x00,0x00,
  //功能描述符
  0x07,0x24,0x01,0x00,0x01,0x25,0x00,
  //IN-JACK
  0x06,0x24,0x02,0x01,0x01,0x00,
  0x06,0x24,0x02,0x02,0x02,0x00,
  //OUT-JACK
  0x09,0x24,0x03,0x01,0x03,0x01,0x02,0x01,0x00,
  0x09,0x24,0x03,0x02,0x04,0x01,0x01,0x01,0x00,
  //以下为两个端点的描述符
  0x07,0x05,0x02,0x02,0x40,0x00,0x00,
  0x05,0x25,0x01,0x01,0x01, /* EMB MIDI JACK = 1, AssocJACKID=1, OUT */
  0x07,0x05,0x82,0x02,0x40,0x00,0x00,
  0x05,0x25,0x01,0x01,0x03 /* EMB MIDI JACK = 1, AssocJACKID=3, IN */
  
};
/*字符串描述符*/
unsigned char  __code LangDes[]={0x04,0x03,0x09,0x04};       //语言描述符
unsigned char  __code SerDes[]={                 // Serial number string descriptor
          0x14,0x03,
          '2',0x00,'0',0x00,'1',0x00,'8',0x00,'-',0x00,
          '3',0x00,'-',0x00,
          '2',0x00,'7',0x00
                 };
unsigned char  __code Prod_Des[]={                // Product string descriptor
          0x14,0x03,
          'C',0x00,'H',0x00,'5',0x00,'5',0x00,'x',0x00,'M',0x00,
          'I',0x00,'D',0x00,'I',0x00,
                 };
unsigned char  __code Manuf_Des[]={
  0x0A,0x03,
  0x5F,0x6c,0xCF,0x82,0x81,0x6c,0x52,0x60,
};
[... 481 lines cut ...]
        #endif
        
        UEP2_CTRL = UEP2_CTRL & ~ MASK_UEP_R_RES | UEP_R_RES_ACK;
        length = USBByteCount;
        USBByteCount = 0;

      }
      if(MidiByteCount)
        Midi_Timeout++;
      if(!UpPoint2_Busy)   //端点不繁忙（空闲后的第一包数据，只用作触发上传）
      {
        //length = MidiByteCount;
        if(length>0)
        {
            //写上传端点
            memcpy(Ep2Buffer+MAX_PACKET_SIZE, Receive_Midi_Buf, length); //环回MIDI设备
            UEP2_T_LEN = length;                //预使用发送长度一定要清空
            UEP2_CTRL = UEP2_CTRL & ~ MASK_UEP_T_RES | UEP_T_RES_ACK;     //应答ACK
            UpPoint2_Busy = 1;
            length = 0;
         }
      }
    }
  }
}

DeqingSun, 2020-10-22:
Did you compile and get it work with SDCC directly?
Did you try to use "USER CODE w/ 148B USB ram" when you compile in Arduino IDE?

NoNamedCat, 2020-10-24:
Yes, i tryied both... I think that the problem is in my "port"

closed 2020-10-25 by DeqingSun