DeqingSun/ch55xduino #17: USB DTR RTS lines in UsbSerialAdaptor example

NoNamedCat, 2020-10-22:
I think that this module would be a great addition to other microcontrollers that not have USB stack. But for some microcontrollers like the ESP8266 or the ESP32 we need those lines.
So.... Here is the code:
```c
/*
  UsbSerialAdaptor
  A simple example turns CH552 to a serial adaptor, 
  relaying data between USB serial and Serial0.
  Baudrate of Serial0 is determined by USB serial's rate.
  created 2020
  by Deqing Sun for use with CH55xduino
  This example code is in the public domain.
*/

#define RTS_LINE 11
#define DTR_LINE 33

extern __xdata uint8_t LineCoding[]; //lineCoding of CDC is located in this array
extern __xdata uint8_t controlLineState;
__xdata uint32_t oldBaudRate = 9600;
__xdata uint32_t oldLineState = 0;

void setup() {
  Serial0_begin(9600);
  pinMode(RTS_LINE, OUTPUT);
  pinMode(DTR_LINE, OUTPUT);
}

void loop() {
  while (USBSerial_available()) {
    char serialChar = USBSerial_read();
    Serial0_write(serialChar);
  }
  while (Serial0_available()) {
    char serialChar = Serial0_read();
    USBSerial_write(serialChar);
  }

  __xdata uint32_t currentBaudRate = *((__xdata uint32_t *)LineCoding); //both linecoding and sdcc are little-endian
  __xdata uint32_t currentLineState = controlLineState; //both linecoding and sdcc are little-endian

  if (oldBaudRate != currentBaudRate) {
    oldBaudRate = currentBaudRate;
    Serial0_begin(currentBaudRate);
  }

  if (oldLineState != currentLineState) {
    oldLineState = currentLineState;
    digitalWrite (RTS_LINE ,((currentLineState & 0x02) == 0x02));
    digitalWrite (DTR_LINE ,((currentLineState & 0x01) == 0x01));
  }
}
```

NoNamedCat, 2020-10-24:
I think that this dont work as expected because when i put the DTR line up, the serial converter stop working. I think its because the bootloader code...
Can i bypass that option in the CDC code to get a working Serial Converter with DRT & RST lines working?

DeqingSun, 2020-12-03:
I can not reproduce the reset issue. In USBCDC.c, the reset will only occur when then baud rate is 1200.

closed 2021-02-12 by DeqingSun