DeqingSun/ch55xduino #23: CMSIS-DAP example fails on Win7

nerdralph, 2020-11-30:
I've tried both with a CH552 and a CH551 module.
After power cycling the module, windows detects it, but under device status, the device status is "device cannot start (Code 10)"
The UsbSerialAdaptor example works fine.

nerdralph, 2020-11-30:
I reverted to v0.0.5, and get the same problem.

DeqingSun, 2020-12-01:
Thanks. I can reproduce the same error on Win7. Will take a look.

DeqingSun, 2020-12-02:
It seems a drive issue with the same PID. 
Try this one
https://github.com/DeqingSun/ch55xduino/tree/playground/ch55xduino/ch55x/libraries/Generic_Examples/examples/05.USB/CMSIS_DAP

nerdralph, 2020-12-04:
That fixes it.  Now it detects as a HID device, and
`openocd -f interface/cmsis-dap.cfg`
recognizes it as a SWD device.

DeqingSun, 2020-12-05:
Good. It has been tested with OPENOCD on Mac in Arduino IDE with ATSAMD21G18. Worked when I select "ATMEL EDBG" as programming. Programmed bootloader on ATSAMD21G18 well.

Also worked with Keil 5 on Win7 with STM32F103C8T6. Both tested in programming and debugging.

closed 2020-12-05 by DeqingSun