DeqingSun/ch55xduino #26: This is not bug. Thank you ^^

thinhx2, 2020-12-15:
[image]

Thank you, working fine

thinhx2, 2020-12-15:
Btw i have a question : possible to port it to nuvoton w78e0 (preinstalled bootloader with press reset button to flash) or stc 8051 (preinstalled bootloader with disconnect vcc and reconnect vcc to flash)?

DeqingSun, 2020-12-15:
@thinhx2 Definitely possible. You may need to customize the tools to fit your tool chain. And if you need to upload the sketch with Arduino. You may need to design a special serial adaptor to power cycle target. If you only need to press reset button, just refer to Arduino Uno circuit.

closed 2020-12-15 by DeqingSun

thinhx2, 2020-12-15:
Thank for suggest. I will look to stcgal for stc