DeqingSun/ch55xduino #27: USB CDC 串口收发信息问题,接收无反应

wukongxuetang, 2020-12-17:
您好,在阿莫论坛看到您的帖子,这段时间也在测试这个小板子,想请教一个问题

调试步骤:
结合usb文件夹内案例和触摸感应例程,想用电脑接收触摸的编号
烧录后电脑可以识别为usb串行设备(COMxx)的方式
然后用一个usb转串口设备(profilic)连接p3.0和p3.1,gnd,使用两个串口助手互发信息模拟发送和接收

结果:
板子对应的cdc串口调试助手发送信息,profilic可以收到信息,profilic发送的信息,CDC对应串口助手没有反应

求教:
usb cdc编程接收板子信息,比如C#直接使用serialPort类,怎么接收?还是usbCDC例程设置问题?

代码:
#ifndef USER_USB_RAM
#error "This example needs to be compiled with a USER USB setting"
#endif

#include "src/userUsbCdc/USBCDC.h"

#include <TouchKey.h>
#define LED_BUILTIN 32

__xdata uint32_t oldBaudRate = 9600;

#define Rec_Long 64
__xdata char recvStr[Rec_Long], serialStr[Rec_Long];
uint8_t recvStrPtr = 0,serialPtr = 0;

void setup() {
  USBInit();
  Serial0_begin(9600);
  pinMode(LED_BUILTIN, OUTPUT);
  TouchKey_begin( (1 << 0) | (1 << 1) | (1 << 2) | (1 << 3) | (1 << 4) | (1 << 5) );
}

void loop() {
  TouchKey_Process();
  uint8_t touchResult = TouchKey_Get();
  if (touchResult) {
    digitalWrite(LED_BUILTIN, HIGH);
  } else {
    digitalWrite(LED_BUILTIN, LOW);
  }
  for (uint8_t i = 0; i < 6; i++) {
    USBSerial_print_s("TIN");
    USBSerial_print_i(i);
    if (touchResult & (1 << i)) {
      USBSerial_print_s(" X ");
    } else {
      USBSerial_print_s(" _ ");
    }
  }
  USBSerial_println();

  while (USBSerial_available()) {
    char serialChar = USBSerial_read();

    if (recvStrPtr < Rec_Long) {
      recvStr[recvStrPtr] = serialChar;
      recvStrPtr++;
    }
  }

  if (recvStrPtr > 0) {
    Serial0_print_s("From UsbCDC:");
    Serial0_println_s(recvStr);
    recvStrPtr = 0;
    memset(recvStr, '\0', sizeof(recvStr));
  }
  
  while (Serial0_available()) {
    char serialChar = Serial0_read();

    if (serialPtr < Rec_Long) {
      serialStr[serialPtr] = serialChar;
      serialPtr++;
    }
  }

  if (serialPtr > 0) {
    USBSerial_print_s("From Serial:");
    USBSerial_println_s(serialStr);
    serialPtr = 0;
    memset(serialStr, '\0', sizeof(serialStr));
  }
}

DeqingSun, 2020-12-17:
Did you try the UsbSerialAdaptor example?

wukongxuetang, 2020-12-18:
Just have a try used the UsbSerialAdaptor example,PL2303 can received USB CDC send message,but usb cdc couldn't received pl2303's?

arduino 1.8.13 win10 x64

看到您回复就进行了测试,烧录后串口设备里可以看到板子的串口信息,用其他串口设备连接进行互发测试,板子的虚拟串口发送的信息,pl2303可以收到,pl2303发送的信息,串口工具中没有反馈?

wukongxuetang, 2020-12-18:
[](url)
[image]
[image]
[image]

DeqingSun, 2020-12-18:
Can not duplicate the error with CH552 and PL2303. Double check if you connection. P30 is RX and P31 is TX. 
Also try to use a jumper to connect P30 an P31 and see if the Ch552 and do an echo test.

wukongxuetang, 2020-12-19:
used a jumper to connect P30 an P31 ,but not found an echo test,I think there's something wrong with my board

closed 2021-02-12 by DeqingSun

wukongxuetang, 2021-02-23:
I found the reason about error,Please use RTS or DTR when test,Thanks