DeqingSun/ch55xduino #28: add 16/24Mhz F_CPU option

nerdralph, 2020-12-20:
I just did some testing with a CH552 running at 3V3, and running at 24MHz is unstable; it seems to run for a few ms after changing the PLL, then it hangs.  At 16MHz, it's fine.  I had been running the MCU at 5V for a few months and never encountered problems with 24MHz.
The datasheet has some contradictory information about what speeds are supported at what voltages, but given the first chip I tested at 3V3 doesn't work at 24MHz, I think it is worth adding the option to choose 16/24Mhz.

DeqingSun, 2020-12-20:
Is this what you are looking for? The millis and micro also work correctly in 16M.
[image]

nerdralph, 2020-12-20:
Yes.  I forgot I had rolled back to v0.5 when testing CMSIS-DAP.

nerdralph, 2020-12-20:
After upgrading to 0.0.7, re-copying the CMSIS-DAP example files from master, and choosing 16MHz clock, pyocd sees the module when it is running at 3V3.

closed 2020-12-20 by nerdralph