DeqingSun/ch55xduino #31: Is there any issue on WS2812?

yanbwang, 2021-01-14:
Hi，

I'm using WS2812B-Mini now. 

The datasheet is below 

https://atta.szlcsc.com/upload/public/pdf/source/20190620/C114583_ECCAEB884D1CBA01F5696FFC90F90D84.pdf

I find there is something wrong with Blue color. Eg. I set RGB as 00,00,FF， the color is Red+Blue.

And I also noticed you have remove the example file of WS2812 in your code.

Could you tell me why? Have you found some issue in your WS2812 library?

Thanks.

Z.t

DeqingSun, 2021-01-14:
I didn't remove the example. It is still on 
https://github.com/DeqingSun/ch55xduino/blob/ch55xduino/ch55xduino/ch55x/libraries/WS2812/examples/chaseTest/chaseTest.ino

I've tested on my WS2812 and it worked well. If doesn't work on your side. Would you mind providing more sample of the wrong color?

yanbwang, 2021-01-15:
Hi，

Yes, I can see your WS2812 code now. I don't know why the ZIP I downloaded didn't have this demo code.

I will study it tonight.

Thanks，

Z.t

yanbwang, 2021-01-16:
Hi,

I have fixed this issue. It's caused by the time requirement is different from the one in library. 

       WS2812D-F8 | WS2812B-Mini
T0H | 400±150ns | 300±80ns
T0L | 850±150ns | 790±210ns
T1H | 850±150ns | 790±210ns
T1L | 400±150ns | 790±210ns
RES | >50us        | >280us

After I change the data transfter time as the one of  WS2812B-Mini, it works well.

Thanks a lot for your works.

Z.t

closed 2021-01-16 by yanbwang

buttim, 2023-11-28:
How did you change the timing? It seems the only possibility here is changing the number of nops in the code.
I have some issues with the last of 16 LEDs. The last LED always has blue on.

DeqingSun, 2023-11-28:
@buttim if you see a lot of nop, you should update the CH55xduino
Now you can change WS2812_DELAY

buttim, 2023-11-28:
That's really good news. How comes I can't see the possibility to upgrade from the IDE?

buttim, 2023-11-28:
Did not add the URL in settings actually