DeqingSun/ch55xduino #32: set bLDO3V3_OFF when 3V3/16M is chosen?

nerdralph, 2021-01-15:
Although it doesn't seem to cause any problems leaving the LDO enabled when VCC and V33 are connected to a 3.3V supply, the datasheet does recommend it be set.  This could be done at the same time as the clock frequency is initialized, while the MCU is in safe mode.

DeqingSun, 2021-01-15:
Good idea. But where did you find the "the datasheet does recommend it be set". I didn't find it. Would you mind pointing where it is?

nerdralph, 2021-01-15:
版本：1E
5.4 bLDO3V3_OFF:
USB电源调整器LDO的禁用控制位：该位为0则允许LDO，可以由5V电源产生3.3V电压用于USB和内部时钟振荡器；为1则禁用LDO，V33引脚必须输入外部的3.3V电源

I could be misinterpreting what it says as a "recommendation".  However I can't think of any good reason to leave it enabled.

DeqingSun, 2021-01-16:
I'm reading the same version, this should means:

USB power regulator LDO disable control bit: This bit is 0 to enable LDO, which can generate 3.3V voltage from 5V power supply for USB and internal clock oscillator; 1 to disable LDO, V33 pin must input external 3.3V power supply

I get it. However, I'm a bit confused how to implement it. What if user is using 5V and 16M. Will the 2 options make things unnecessary complicated?

nerdralph, 2021-01-16:
You could have 2 separate menu options, one for clock speed (16 or 24 MHz), and another menu for LDO disabled for VDD=3V3.  I agree that the more options you add, the more confusing it is for users.  Perhaps if your priority is simplicity for the users, then adding an extra menu option isn't justified.

closed 2022-08-12 by DeqingSun