DeqingSun/ch55xduino #33: Uart0 can't work with 115200 baud rate after chosen clock source as 16Mh (internal), 3.3v or 5V

yanbwang, 2021-01-31:
Hi,

I find my board(CH554, 3.3V power) Uart0 does not work when the baudrate is set as 115200 and clock source as "16Mh (internal), 3.3v or 5V." But it works well with 9600 with "clock source as 16Mh (internal), 3.3v or 5". And I use another board(CH552) with 5V power, it has same issue. In this way, I confirm it's not caused by power.

I use a FT232 USB to serial convertor as a sender. Its TXD is connected to P3.0(Pin10). And here is a simple code which will send the data got from Uart0 to USB port.

`
#ifndef USER_USB_RAM
#error "This example needs to be compiled with a USER USB setting"
#endif

#include "src/userUsbCdc/USBCDC.h"

void setup() {
  USBInit();
    Serial0_begin(115200);
}

void loop() {
  while (Serial0_available()) {
    char serialChar = Serial0_read();
    USBSerial_write(serialChar);
    USBSerial_flush();
  }

}`

Could you please check it?

Thanks.

Z.t

DeqingSun, 2021-01-31:
That is a know issue. CH552 can not communicate properly with 115200 baudrate on 16Mhz. The Uart clock pre-scaler is 16, and the closest divider we can use is 9, which sets baud rate to 111111. And that is a 3.5% difference from 115200 and cause the communication failure.

You may either use a lower rate or non-standard baudrate. I think FT232 do support non-standard baudrate such as 100000.

yanbwang, 2021-02-01:
Thanks a lot. I will try it with 100000.

closed 2021-02-01 by yanbwang