DeqingSun/ch55xduino #35: I cannot write from the ArduinoIDE.

ichirowo, 2021-02-10:
When you press the Write button, the ArduinoIDE will display an error.

------------------------------------------------------------------

CH55x Programmer by VNPro

------------------------------------------------------------------

Load file as hex

   Loaded 4388 bytes between: 0000 to 1151

Found no CH55x USB
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

[image]

[image]

Are the settings correct? Is there anything else I can do?

prosper00, 2021-02-10:
If it's been flashed before, you need to put it into DFU mode. Typically, you'd do this by disconnecting the device, connecting USB Data+ to 3.3V with a 10k resistor, and then powering it up. (you can remove the resistor at this point). That should put it in DFU mode, and you can upload to it again.

ichirowo, 2021-02-10:
The previous image shows DFU mode.
[image]

It just looks like this when you turn it on
[image]

ichirowo, 2021-02-10:
You don't need to configure the COM port, do you?
It is not recognized as a COM port in DFU mode of CH552

[image]

prosper00, 2021-02-10:
I've never used the com port/CDC option, so I can't help you with that.

I've also noticed that I have to start programming very soon after entering DFU. If I enter DFU and leave it for more than a few seconds before I start programming, then it doesn't detect. I don't know why. It still shows as connected in DFU mode, but, programming doesn't work. However, if I start programming within a few seconds of connecting the device, then it all works

DeqingSun, 2021-02-10:
@ichirowo If you see "USB module". It is likely the DFU mode is correct. It is likely a driver issue and the uploader can not communicate with hardware.

Did you use the Zadig software to install the driver? You should see something like this.

[image]

ichirowo, 2021-02-11:
Thanks. It's resolved.

I didn't think it was important to install the driver in Zadig.

When I couldn't write, I had the following problem.
[image]

When I did the following, I was able to write with ArduinoIDE.
[image]

[image]

I could also confirm the symptom of not being able to write after about 10 seconds of entering DFU mode.
You should write as soon as you switch to it.

Thanks. I will introduce this great product and community to Japan.

DeqingSun, 2021-02-11:
@ichirowo Great. If you use the default CDC setting, choose the correct COM port and the CDC code will enter DFU automatically when you upload the code.

ichirowo, 2021-02-12:
Thanks. It's working fine.
I also installed the CDC driver.
[image]

DeqingSun, 2021-03-07:
Seems you get it solved?

hexesdesu, 2021-03-11:
I've got the same problem. And once I upload a stretch, then I have to ①unplug the USB connector ②reconnect the connector with BL button pushed ③find the device which has a yellow alert sign in device manager and choose automatically update device driver ④reinstall WinUSB driver in Zadig; **Everytime！**
>I f your board is never used with ch55xduino before, you need to make the ch55x chip enter bootloader mode. You need to disconnect USB and unpower ch55x, connect the pull-up resistor on D+ line (generally a 10K resistor between D+ and 5V, controlled by a push-button or adjacent pads). Then you connect USB. and hit Upload. Also, a blank new chip will enter the bootloader automatically.
> If you have used ch55xduino once and your code doesn't crash the USB subsystem, you can simply press Upload. Arduino and the firmware will kick the chip into the bootloader automatically.
Quite not same/(ㄒoㄒ)/~~

DeqingSun, 2021-03-11:
@hexesdesu what version of Windows are you using? Also would you mind sharing a screen shot of the yellow alert sign, and the details (error code, etc) in that device? That will help me to troubleshoot it.

SdtElectronics, 2021-03-12:
I have met the same issue as the one mentioned by @hexesdesu , however, I haven't solved it yet. My current Windows installation is 2004 (internal version 19041.804) but I have already encountered this issue for a long time, likely in version 190x. 
The error reported is "Device USB\VID_4348&PID_55E0\5&92c31b4&0&9 had a problem starting" with error code 0xA. This only happens after I installed the WinUSB driver with Zadig. With the original driver provided by wch, the device can be recognized by the system.

On the other hand, the official flashing tool seems worked nicely with the original driver. I wonder whether there is an alternative way to download the bin generated by the arduino IDE, with the official flashing tool?

DeqingSun, 2021-03-12:
@SdtElectronics I think there might be a conflict with the official driver, as the official driver is listed on Windows Hardware Compatibility List. I'll see if I can use the CH372 API (official one) in windows to solve this issue.

Before that, maybe you can try to remove the official driver completely and then use the Zadig, to see if it works.

The process should be similar to this one (remove the official driver, not your wireless card. This image is from internet). 

[image]

hexesdesu, 2021-03-15:
I think I have fix this problem by using libusb instead of Winusb Driver

**Winusb**
first time plugged in

[image]

second time plugged in

[image]

**libusb**
first time plugged in

[image]

second time plugged in ( •̀ ω •́ )y

[image]

hexesdesu, 2021-03-15:
@SdtElectronics Maybe have a try?

SdtElectronics, 2021-03-16:
@hexesdesu Whoa! This worked for me! That cursed WinUSB has bothered me for several months. I am really grateful to your help.

DeqingSun, 2021-03-16:
@hexesdesu Thanks for the help, I'll update the readme

DeqingSun, 2021-03-20:
[vnproch55x.zip](https://github.com/DeqingSun/ch55xduino/files/6174738/vnproch55x.zip)
@HonghongLu Please try this file. It has libusb statically linked. Replace the original file in the Arduino package. On my computer, it was in ~/Library/Arduino15/packages/CH55xDuino/tools/MCS51Tools/2020.12.01/macosx 

Note you may need to allow the app in the security setting.

HonghongLu, 2021-03-20:
> [vnproch55x.zip](https://github.com/DeqingSun/ch55xduino/files/6174738/vnproch55x.zip)
> @HonghongLu Please try this file. It has libusb statically linked. Replace the original file in the Arduino [...]

yes, I replaced and file and tried. here is another error. 
------------------------------------------------------------------
CH55x Programmer by VNPro
------------------------------------------------------------------
Load file as hex
   Loaded 4393 bytes between: 0000 to 1156
DeviceVersion of CH55x: 1.00 
MCU ID: 00 00
Not support, family ID.
An error occurred while uploading the sketch

DeqingSun, 2021-03-20:
@HonghongLu Unfortunately my tool does not support the old protocol. All chip I can get have a 2.3.1 or newer boot loader version. You can try with a new chip. Or you can use the upload tool from other people.

AnonymousChatter, 2021-05-29:
@DeqingSun Just would like to chime in on this topic. I have been testing out the WebUSB uploading tool & Chrome would only support WinUSB, so LibUSB option is out. 
It works flawlessly with Windows 7 using Zadig's WinUSB driver. 
But for Windows10, it work once after installing as observe  by @hexesdesu , subsequent DFU mode would end up with a non operation device + driver.
Here is the funny part. If I forcibly use Zadig to use WinCDC with the DFU CH55x (Of course it will fail), then swap the driver back to WinUSB again, it works again :)

DeqingSun, 2022-03-31:
@nulllaborg You can use the CH375 driver now
https://github.com/DeqingSun/ch55xduino/blob/80d298d92efb4f4f9473f86be5710ac8f1b719b4/README.md?plain=1#L56

closed 2022-03-31 by DeqingSun