DeqingSun/ch55xduino #36: hard-coded Ep0 size bug

nerdralph, 2021-02-12:
https://github.com/DeqingSun/ch55xduino/blob/ch55xduino/ch55xduino/ch55x/libraries/Generic_Examples/examples/05.USB/CMSIS_DAP/src/CMSIS_DAPusb/USBhandler.c#L9

If DEFAULT_ENDP0_SIZE is changed to 64, the code will break.

DeqingSun, 2021-02-13:
You may also need to change EP1_ADDR, EP2_ADDR, etc, and USER_USB_RAM. Or there may be collision on those arrays.

nerdralph, 2021-02-13:
I've already started changing the code so the buffers are allocated by the
compiler instead of using fixed addresses.

DeqingSun, 2021-02-13:
The official example mentioned the DMA arrays must allocated in even address in comment. Although I didn't find such thing in the datasheet. I know there is a .even keyword for alignment in Flash but I didn't check if there is way to do it in RAM. If you finds such a way, please let me know. It can simplify the menu as well.

nerdralph, 2021-02-13:
I can do it when I ensure each object file has xmem aligned and padded to
an even number of bytes.  Its easy for a project with a few files, but
would be harder with ch55xduino.
https://sourceforge.net/p/sdcc/mailman/message/37218815/
Maybe with a custom linker script you could do it.

closed 2022-08-12 by DeqingSun