DeqingSun/ch55xduino #39: How to get ch55x's Unique ID

wukongxuetang, 2021-02-23:
how to get ch55x's Unique ID?

DeqingSun, 2021-02-23:
```
  USBSerial_print_ub(*(__code uint8_t*)(0x3FFC), HEX);
  USBSerial_print_c(' ');
  USBSerial_print_ub(*(__code uint8_t*)(0x3FFD), HEX);
  USBSerial_print_c(' ');
  USBSerial_print_ub(*(__code uint8_t*)(0x3FFE), HEX);
  USBSerial_print_c(' ');
  USBSerial_print_ub(*(__code uint8_t*)(0x3FFF), HEX);
  USBSerial_print_c(' ');
  USBSerial_print_ub(*(__code uint8_t*)(0x3FFA), HEX);
  USBSerial_print_c(' ');
  USBSerial_print_ub(*(__code uint8_t*)(0x3FFB), HEX);
  USBSerial_print_c(' ');
  USBSerial_println();
```

I get

```
3 C0 BC 4C 31 52
```

nerdralph, 2021-02-24:
0x3FFB isn't part of the unique ID.  The datasheet says it is reserved "保留", though it seems to be used for the chip ID, just like SFR CHIP_ID at address 0xA1.  With a CH551, this should output 51.

wukongxuetang, 2021-02-24:
I am using ch552,Test the code,I get 1C 7E 80 58 31 52. @nerdralph  0x3FFB value is same with CHIP_ID.

wukongxuetang, 2021-02-27:
I am using ch552,Test the code,I get 1C 7E 80 58 31 52;
I test another ch552g,get uid is 03 7E 80 58 55 5A,0x3FFB value is 5A,not same with chip_id,

closed 2022-08-12 by DeqingSun