DeqingSun/ch55xduino #42: Timers with the Arduino IDE

Vykynger, 2021-03-05:
Hey,
I am facing some trouble with the Timers in the Arduino IDE. When working with the [CH55xTool](https://github.com/MarsTechHAN/ch552tool) my Code works, but with the Arduino IDE, it does not. However, I would like to use some of your libraries for example for printing into the Serial Monitor. Below there is an example code, that I could not get to run. It would be really nice if you could tell me what the problem is.
`
#define LED_PIN 7

SBIT(LED, 0x90, LED_PIN);
int counter = 0;
/**
SFR(TMOD,  0x89);  // timer 0/1 mode
SFR(TH0,  0x8C);  // high byte of timer 0 count
SFR(TL0,  0x8A);  // low byte of timer 0 count
SBIT(TR0,  0x88, 4); // timer0 run enable
SBIT(ET0,  0xA8, 1); // enable timer0 interrupt
SBIT(EA,  0xA8, 7); // enable global interrupts: 0=disable, 1=enable if E_DIS=0
**/
void setup() {
    //CfgFsys(); <-- from ch554.h
    // Configure pin 1.7 as GPIO output
    P1_DIR_PU &= 0x0C;
    P1_MOD_OC = P1_MOD_OC & ~(1<<LED_PIN);
    P1_DIR_PU = P1_DIR_PU |  (1<<LED_PIN);

      // Init
   TMOD = 0x11;       // set Timers to 16-Bit Mode
   TH0 = 0;           // Reset High-Byte
   TL0 = 0;           // Reset Low-Byte
   TR0 = 1;          
                      // wird jetzt TH0:TL0 um 1 hochgezählt
   ET0 = 1;           // Activate Timer 0 Interrupt
   EA  = 1;           //Activate Globle Interrupt

}
void loop()
{
  //do something with the counter
}
#pragma NOOVERLAY
void Timer0(void) __interrupt (INT_NO_TMR0)
{
        counter = counter + 1;
        LED = !LED;
        TH0 = 0;  // reset of High-Byte
        TL0 = 0;  // reset of Low-Byte
}`

DeqingSun, 2021-03-05:
If you check wiring.c you will see the timer0 is reserved for  millis and micros function for timing. This is a common practice for Arduino. You can use T2 or T1 if you don't use the uart module.

Vykynger, 2021-03-06:
Actually, it seems like I am configuring the Timers wrong because Timer 1 and external Interrupt 1 are not working either. Unfortunately, I did not find any example of a CH55x Timer implementation using the Arduino IDE. Do you know where to find a working example?  Also, I did not quite understand the `#pragma NOOVERLAY` thing because the Arduino IDE does not seem to recognise it.

Vykynger, 2021-03-06:
But thanks a lot for the hint with Timer 0. I would have never thought about that. :)

Vykynger, 2021-03-06:
Maybe it is one of those issues: 
_If you want to use interrupt. You need to add handler both in your code and main.c._
What do you mean with this sentence? "handler" = Interrupt Service Routine? But what is main.c? I only have the blabla.ino file when coding with the Arduino IDE. And what do you mean with code? I obviously have the handler in my code, as seen above. :dagger:
_And using direct register acesss is generally not recommended for Arduino, unless the library can not do what you want_ 
How can I configure the timers then? I am not that proficient in C.

DeqingSun, 2021-03-06:
Those existing isr handlers are for External int 0 and 1, and timer 9

Vykynger, 2021-03-06:
Ok thanks for that fast answer. But how do I set up an external Interrupt then? What ist intFunc[1]? Again writing an normal ISR that handles te external interrupt in the Arduino file does not seem to work. Also naming the ISR in the Arduino Code INT1_ISR(void) does not work either. Can you maybe just give me an example or tell me a little bit detailed on how to use the Ext. Interrupt with the Arduino IDE? 
Below this there is an example that I would like to get to run using the Arduino IDE.:
```
#include <ch554.h>
#include <debug.h>

#define LED_PIN 7
SBIT(LED, 0x90, LED_PIN);
void main() {
    CfgFsys();
    P1_DIR_PU &= 0x0C;
    P1_MOD_OC = P1_MOD_OC & ~(1<<LED_PIN);
    P1_DIR_PU = P1_DIR_PU |	(1<<LED_PIN);

      // Initialisierung
   IT1 = 1;    // Falling-Edge Trigger for Ext. Interrupt 1
   EX1 = 1;    // External Inteerupt 1 aktivieren
   EA  = 1;    // Globalen Interrupt aktivieren
   while (1) 
   {
   }
}
void Ext(void) __interrupt (INT_NO_INT1)
{
        LED = !LED;
}
```

DeqingSun, 2021-03-06:
Check https://github.com/Blinkinlabs/ch554_sdcc/blob/3b86045fbefc1ff38484acbfa4289cf923451033/include/ch554.h#L695 for info about adding additional ISR in main.c. That will give you a hint about how to add another ISR.

Back to you external ISR request, check https://github.com/DeqingSun/ch55xduino/blob/ch55xduino/ch55xduino/ch55x/libraries/Generic_Examples/examples/02.Digital/AttachInterrupts/AttachInterrupts.ino

If you have never played with any Arduino board, try the examples to understand how the Arduino works.

Vykynger, 2021-03-07:
Thank you very very much for your help!  I had not found this example library yet. But with it is really easy to program a wch552 with the Arduino IDE. :)

closed 2021-03-07 by Vykynger