DeqingSun/ch55xduino #43: shift registers problem

sh4had, 2021-03-28:
I wanted to use a CH552 and a seven-segment display using a 74hc595 shift registers. 
On stm8s103f3p6 (sduino) it works fine, but ch552 doesn't even compile.
I get errors:
```
warning 112: function 'shiftOut' implicit declaration
warning 112: function 'shiftOut' implicit declaration
error 101: too many parameters 
error 101: too many parameters 
```
[image]

If I drop these two lines with "shiftOut", the sketch will compile successfully.

DeqingSun, 2021-03-28:
The shiftin and shiftout is not implemented yet. You can use a for loop do it.

DeqingSun, 2021-03-29:
@sh4had you may add this function in your sketch to use  the shiftout function.

```
void shiftOut(uint8_t dataPin, uint8_t clockPin, uint8_t bitOrder, uint8_t val)
{
 uint8_t i;
 for (i = 0; i < 8; i++) {
 if (bitOrder == LSBFIRST)
 digitalWrite(dataPin, !!(val & (1 << i)));
 else
 digitalWrite(dataPin, !!(val & (1 << (7 - i))));
 digitalWrite(clockPin, HIGH);
 digitalWrite(clockPin, LOW);
 }
}

``

sh4had, 2021-03-29:
Thanks! Now everything works.

closed 2021-03-29 by sh4had