DeqingSun/ch55xduino #44: CH552T upload problems

hexesdesu, 2021-03-30:
When I tried to upload a blink demo to a CH552T chip, I got:
```
------------------------------------------------------------------
CH55x Programmer by VNPro
------------------------------------------------------------------
Load file as hex
   Loaded 4393 bytes between: 0000 to 1156
Not support
```

[image]
(failed to upload the project)

But with the same code, CH551G and CH522G works just fine.

Then I export the hex file and upload it by WCHISPTool, and it works.

Maybe it's a VNPro compatible problem？

DeqingSun, 2021-03-30:
@hexesdesu Are you using an early version of ch55xduino? Please update to the newest version and try again. It may work or it will show you more info to troubleshoot.

hexesdesu, 2021-03-31:
> @hexesdesu Are you using an early version of ch55xduino? Please update to the newest version and try again. It may work or it will show you more info to troubleshoot.

Well, it's the newest 0.0.9 version downloaded by Arduino board manager.
You've inspired me and I will try a lower version.

DeqingSun, 2021-03-31:
@DeqingSun 

That is quite weird, I tried it again in windows, and it shows the bootloader ID and chip ID. Even there is a error message, the ID should be printed first.

Maybe you can delete the board support file completely, include the caching zip files and install again.

[image]

hexesdesu, 2021-03-31:
> @DeqingSun
> 
> That is quite weird, I tried it again in windows, and it shows the bootloader ID and chip ID. Even there is a error message, the ID should be printed first.
> 
> Also, the newer upload tool will not just say "not support". It will give you a reason.
> 
> Maybe you can delete the board support file completely, include the caching zip files and install again.
> 
> [image]

Yep I tried CH551G and CH552G and they do show the bootloader ID and chip ID, only my CH552T not.

I tried to install Arduino in a new PC and I got the same.

I got these chips from Xianyu, which the seller claimed that they are shipped by szlcsc. And it can be upload by WCHISPTool, so I don't think it was a chip failure.

I bought some new chips from Taobao in order to figure how this could happened.

DeqingSun, 2021-03-31:
That really weird. I do have some CH552T and there is no difference from the CH552G at all. 
And the new tool has no string "Not support". Such a string only exist in the old version of tools.
By the way, can you upload hex with https://deqingsun.github.io/ch55xduino/bootloaderWebtool/ ?

hexesdesu, 2021-03-31:
> That really weird. I do have some CH552T and there is no difference from the CH552G at all.
> And the new tool has no string "Not support". Such a string only exist in the old version of tools.
> By the way, can you upload hex with https://deqingsun.github.io/ch55xduino/bootloaderWebtool/ ?

It says `MCU family Not support ` :(

hexesdesu, 2021-03-31:
> That really weird. I do have some CH552T and there is no difference from the CH552G at all.
> And the new tool has no string "Not support". Such a string only exist in the old version of tools.
> By the way, can you upload hex with https://deqingsun.github.io/ch55xduino/bootloaderWebtool/ ?

modified `ch55xbl.js`
```
    await device.transferOut(endpointOut, (new Uint8Array(bootloaderDetectCmd)).buffer)
    //await device.transferOut(endpointOut, (new Uint8Array(bootloaderIDCmd)).buffer)
    result = await device.transferIn(endpointIn, 64);  
    resultUint8 = new Uint8Array(result.data.buffer)
    bootloaderDeviceID = resultUint8[4]
    console.log(result.data.buffer);
    console.log(resultUint8[4]);
    console.log(resultUint8[5]);
```

[image]

 And got definitely creepy response...

DeqingSun, 2021-03-31:
@hexesdesu 
I check another uploader repo https://github.com/ole00/chprog . It seems the V1 version of the bootloader returns 2 bytes instead of 2. Since you get the chip from a flee market. It is likely you get a old chip with old bootloader. 

I never get a chance to get the old version bootloader to test, so the upload tool in my repo don't support that version. You may get a newer chip or use a tool support V1 version of the bootloader.

hexesdesu, 2021-04-01:
> @hexesdesu
> I check another uploader repo https://github.com/ole00/chprog . It seems the V1 version of the bootloader returns 2 bytes instead of 6. Since you get the chip from a flee market. It is likely you get a old chip with old bootloader.
> 
> I never get a chance to get the old version bootloader to test, so the upload tool in my repo don't support that version. You may get a newer chip or use a tool support V1 version of the bootloader.

Ok I checked my bootloader version in WCHISPTool
CH552T:
[image]
CH552G:
[image]
and it seem to be in a lower bootloader version(1.0)

I'm gonna try if I can upgrade it by myself, or bypass the code check.
Thank you for your works on my problems. glhf

DeqingSun, 2021-04-01:
You can not bypass the code check in my uploader, because the protocols are different. I would recommend using another tool to upload.

hexesdesu, 2021-04-01:
Maybe I can use chprog instead of vnproch551 in arduino?

DeqingSun, 2021-04-01:
If you can make it accept hex file, yes.
Refering to https://github.com/DeqingSun/vnproch551/blob/master/KT_BinIO.cpp for hex file parsing.
And if you successfully get it working. Share the solution to help other people.

hexesdesu, 2021-04-02:
Ye I'm gonna try. It's my first time hacking Arduino so it may take a long time.

DeqingSun, 2021-04-02:
You don't need to hack Arduino, just make the chprog to accept the same parameters to the vnproch551, and replace the one in the tool folder.