DeqingSun/ch55xduino #45: Uploading a code via UART

radoslavilievm2m, 2021-04-02:
Hi, I have to have the possibility to upload a code via UART.
Does do you mean enter in bootload mode to access as a function 0xF400 address of the CH559 chip or something else? I tried it in this way, but without any success.
Thank you in advance.

tapir, 2021-06-08:
I would also appreciate if you could shed some light on how to upload data via UART. Do we still need bootloader in this case, can just override it?

radoslavilievm2m, 2021-06-08:
Hi,
The vendor of CH559 chips have an application, that do it on UART0 0.2 and 0.3 pins without modifying the bootloader, but the protocol is closed source.
You can do IAP (In-app programing) by your own about it, and after several emails with WCH they told me that this is the best variant for doing it.

tapir, 2021-06-08:
OK so looks like the USB programming is the way to go then. Thanks.
For IAP there is some code examples here: https://github.com/SoCXin/CH559/tree/master/tools/IAP

DeqingSun, 2021-06-08:
The upload tool does support serial upload. At least for ch554. Other variant not tested.

https://github.com/DeqingSun/vnproch551

The binary can be found in the Arduino package. But you need to run it in command line.

radoslavilievm2m, 2021-06-08:
Okay, I'll test it when I have the possibility and write the result. I hope it will work, because it will fix a lot of problems around my project.

closed 2022-08-12 by DeqingSun