DeqingSun/ch55xduino #48: DTR Line problem

NoNamedCat, 2021-04-19:
`/*
  UsbSerialAdaptor
  A simple example turns CH552 to a serial adaptor, 
  relaying data between USB serial and Serial0.
  Baudrate of Serial0 is determined by USB serial's rate.
  created 2020
  by Deqing Sun for use with CH55xduino
  This example code is in the public domain.
*/

#define RTS_LINE 11
#define DTR_LINE 33

extern __xdata uint8_t LineCoding[]; //lineCoding of CDC is located in this array
extern __xdata uint8_t controlLineState;
__xdata uint32_t oldBaudRate = 9600;
__xdata uint8_t oldLineState = 1;

void setup() {
  Serial0_begin(9600);
  pinMode(RTS_LINE, OUTPUT);
  pinMode(DTR_LINE, OUTPUT);
}

void loop() {
  while (USBSerial_available()) {
    char serialChar = USBSerial_read();
    Serial0_write(serialChar);
  }
  while (Serial0_available()) {
    char serialChar = Serial0_read();
    USBSerial_write(serialChar);
  }

  __xdata uint32_t currentBaudRate = *((__xdata uint32_t *)LineCoding); //both linecoding and sdcc are little-endian
  __xdata uint8_t currentLineState = controlLineState; //both linecoding and sdcc are little-endian

  if (oldBaudRate != currentBaudRate) {
    oldBaudRate = currentBaudRate;
    Serial0_begin(currentBaudRate);
  }

  if (oldLineState != currentLineState) {
    oldLineState = currentLineState;
    digitalWrite (RTS_LINE ,!((currentLineState & 0x02) == 0x02));
    digitalWrite (DTR_LINE ,!((currentLineState & 0x01) == 0x01));
  }
}`

When DTR line go UP the microcontroller stop writing to the serial console. Im trying to use this microcontroller as a serial programmer for the ESP8266 and ESP 32.

I think that the problem is in the file USBCDC.c

DeqingSun, 2021-04-28:
USBCDC.c doesn't care DTR line, it is likely be stopped by your computer driver.

NoNamedCat, 2021-04-29:
I have a linux distro on my computer. With others MCUs like the ATMEL-32u4 i dont have this issue. How can i debug this? ThankU for the help

DeqingSun, 2021-04-29:
I would suggest using a usb analyzer or logic analyzer to capture usb packages to see if the data you want really reached the MCU.
Or if you just have a regular logic analyzer, you can print debug info with sendCharDebug function to see what happens in each usb transaction.

I remember when I test the stm32duino with Android, it has a similar control line issue. That's why I guess it may be an driver issue.

prosper00, 2021-04-29:
While we're on the subject, I've been looking into USB analyzers. Is there one you can recommend? I have no budget for a High Speed/480Mbps one, but I could use a Full Speed/12Mbps unit.

Or, are you referring to software, like wireshark?

DeqingSun, 2021-04-29:
@prosper00 I would recommend basic version of saleae logic 8 with Enthusiast Pricing. The analog feature is quite useful. Or you may choose one on https://sigrok.org/wiki/Supported_hardware#Logic_analyzers or even a clone of those on the list.

prosper00, 2021-04-29:
Hm - I didn't think those would be fast enough to do usb, but I DO have a small salae clone. I'll give it a try.

I really like those little analyzers - having basically unlimited capture memory is super handy. My Rigol scope can only capture for a very short time before it's buffer is full, so I have to spend a ton of time finicking with triggers to try to get the data I want.... and it doesn't have USB decoders or triggers

DeqingSun, 2021-04-29:
@prosper00 I'm not talking about single cypress chip saleae unit. The new saleae logic 8 can do 100M digital and 10M analog.

Or you can take a look at http://www.qdkingst.com/en . They used to clone saleae product but now have their own product lines. I've never used their products but the specs seems good at low price.

prosper00, 2021-04-29:
Thanks! I just ordered a DSLogic Plus (https://www.dreamsourcelab.com/shop/logic-analyzer/dslogic-plus/)
It appears to have better compatibility with Pulseview than the Kingst.

It doesn't do analog, but, that's probably okay as I already have an oscilloscope. And it's half the price of the Salae, even with the discount.

Sorry to OP for sidelining your issue :)

NoNamedCat, 2021-05-28:
Hi... I just want to know... In my serial terminal in the PC, CTS line (for flow control) is up, not like the ch340, or other serial adapters... That's the reason I can't get the messages from the ch551 when I put the DTR line up? How can I change the CTS level in the code?

DeqingSun, 2021-05-28:
The code does not care CTS

NoNamedCat, 2021-05-28:
But how I can change the CTS level? Is there a way to change it?

closed 2022-08-12 by DeqingSun