DeqingSun/ch55xduino #54: Use type-generic macros

leo60228, 2021-09-29:
According to the README:
> There is no free C++ compiler for MCS51 chip, we can not use polymorph functions. So you can not expect the compiler will choose a function according to the parameter's type.

C11 adds [`_Generic`](https://en.cppreference.com/w/c/language/generic) for this purpose. Here's a trivial example: https://godbolt.org/z/Yv8sznf3v

However, this currently can't be used for `print` due to an [SDCC bug](https://sourceforge.net/p/sdcc/bugs/3255/).

DeqingSun, 2021-09-29:
@leo60228 thanks for this info. I'll check if this can be implemented into the package.

DeqingSun, 2021-09-29:
@leo60228 Frankly speaking I have no idea what "_Generic" means or how it compiles (yet). But it seems work. I'll see how to add it into the package.

[image]

leo60228, 2021-09-29:
_Generic is a relatively new feature (C99 had polymorphic functions in the standard library, but those were implemented in a compiler-specific way until C11). SDCC supports it but there's several open bugs. I don't think it's usable for `print` yet, but it could be with some bugfixes, and might be useful for other functions.

DeqingSun, 2021-09-29:
@leo60228 Sorry I ask the question before I learn some basics about  _Generic. Why you think it is not usable for print? It seems my sketch compiles and works as expected.

DeqingSun, 2021-09-29:
@leo60228 Not sure why it works on my side. I'm using a pre-built mac build.11630 from the official website. 

[image]

DeqingSun, 2021-09-29:
@leo60228 Thank you and I get it when trying to use the pointer.

```
Internal error: validateLink failed in SPEC_NOUN(type) @ /Users/sdcc-builder/build/sdcc-build/orig/sdcc/src/SDCCast.c:5401: expected SPECIFIER, got DECLARATOR
```

I'll subscribe to that ticket and see when it get fixed. Not supporting pointer makes implementing the  _Generic to print pointless.

DeqingSun, 2022-03-31:
Update: The bug fixing is now funded. Wish it get fixed soon

https://sourceforge.net/p/sdcc/wiki/SDCC-STD-UX/

DeqingSun, 2022-04-23:
Generic is functional after SDCC commit 13402. 

This repo now supports Generic for print.

https://github.com/DeqingSun/ch55xduino/blob/ch55xduino/ch55xduino/ch55x/cores/ch55xduino/genericPrintSelection.h

closed 2022-04-23 by DeqingSun