DeqingSun/ch55xduino #57: Any USB Host Example

DevDavisNunez, 2021-12-01:
Hello, I had heard that this IC is compatible with USB Host, I am interested in carrying a custom USB protocol that basically only reads and writes some values to a device, can it be done with this IC?

DeqingSun, 2021-12-03:
Yes it is possible. Here is my code to read data from a WS1361 sound meter and forward the data to serial on CH549.
[CH549Host.zip](https://github.com/DeqingSun/ch55xduino/files/7646441/CH549Host.zip)
This code is not well organized to be an example but it does the job. Please refer to the WCH official examples to figure out details about how it works.

From my experience, using an external crystal is necessary for USB host applications.

DevDavisNunez, 2021-12-05:
Should it work with the CH559 ?, I found a very nice evaluation card, I hope it works, as soon as I get to test.

DeqingSun, 2021-12-05:
It should work. But you may need to check if there is any register difference.

closed 2022-08-12 by DeqingSun