DeqingSun/ch55xduino #58: Possible to set Serial String Descriptor using CH552 Unique Serial number?

Kaitoukid93, 2021-12-04:
I have small amount of Products that used CH552 and they worked fine
When my customer purchase more than one for single computer, the COM port is not fixed for each one of ch552 device since they got the same Serial String Descriptor
Is there any way to change Serial String Descriptor using user code ( get 32bit number then set the Serial String Descriptor) so I dont have to complie multiple binary for each device
thanks

DeqingSun, 2021-12-05:
That would require some change in USBconstant.c . You will need to change the code here. https://github.com/DeqingSun/ch55xduino/blob/78d01729a897dd2080c855575f0042dc18c31fc4/ch55xduino/ch55x/cores/ch55xduino/USBhandler.c#L109

However, I made the code more efficient to fix the pDescr to code space. If you also want it to point it to ram space, you will also need to change the type the pDescr to a universal pointer. This will require some work in the code and reduce the code speed.

Kaitoukid93, 2021-12-05:
[image]
Got it! thanks
I don't know about performance loss but the device worked just fine, can you tell me more about it? thanks!

DeqingSun, 2021-12-06:
The generic pointer is 3 bytes. Every time you try to use the pointer, the compiler will call a sub routine to check whether the pointer is in iram, xram, or code space. So the data access is not efficient at all.

Kaitoukid93, 2021-12-06:
[image]
I fix the pDescr to xram
This cost more ram space after compiling

closed 2021-12-10 by Kaitoukid93