DeqingSun/ch55xduino #59: Can I send media control keypresses via HID Keyboard?

foadyousefi, 2021-12-15:
First, thanks for this awesome work.

I am able to compile and flash the examples, but I was wondering how can I send media control commands/keypresses. 
Where can I find a list of available key commands?

The price and minimal circuit requirements of this microcontroller make it ideal for most DIY projects, especially something like macro pads. Setting up a ch552 is way way easier than something like Atmega32u4.

DeqingSun, 2021-12-16:
You do need to make some modifications to the example. Such as adding consumer control in ReportDesc. It will definitely work.

foadyousefi, 2021-12-17:
Thanks for your response. 
Sorry for my dumb question, but can you please give me an example?

DeqingSun, 2021-12-19:
I will try to add that but it may take a few months until I have time to do so.

foadyousefi, 2022-03-18:
@DeqingSun Any news here?

DeqingSun, 2022-03-20:
@foadyousefi 
Try
https://github.com/DeqingSun/ch55xduino/tree/48ab63be79ea0ff72d5b5bb04ace5cf76b6a588a/ch55xduino/ch55x/libraries/Generic_Examples/examples/05.USB/HidMediaKeyboard

foadyousefi, 2022-03-21:
@DeqingSun Thanks, I will try and let you know the results.

foadyousefi, 2022-05-15:
@DeqingSun I tried and it's working perfectly. Thank you.

closed 2022-05-15 by foadyousefi