DeqingSun/ch55xduino #60: CH549 support?

fanoush, 2021-12-18:
H, just to let you know that the cheap $3 CMSIS-DAP dongles on aliexpress like https://www.aliexpress.com/item/1005002820752025.html or https://www.aliexpress.com/item/1005003059548220.html are based on CH549 chip, according to info http://www.wch-ic.com/products/CH549.html it is better CH554 - has more RAM and USB host mode.

I got two those dongles and switched the to bootloader by pulling D+ to 5V and wchisptool detects them both as Ch549. opensource tools shows them as unknown device 0x73 with bootloader 2.40

DeqingSun, 2021-12-19:
It does work with ch549, just some features are not implemented yet.

There isn't a good way to extract firmware. However, WCH has a tool called wch-link. And the schematic and hex file are available. You can use that firmware if the schematic matches your hardware.

fanoush, 2021-12-19:
Thank you for the hint. I found the schematics here https://github.com/kaidegit/CMSIS-DAPbyWCH/blob/main/documents/official%20schematic.pdf and my dongle is wired in same way - leds,uart,swd (only resistors are 220ohm instead of 270, leds are 1k), also the jumper with D+ and pin wired via 4.7k to V33 is there on the bottom of the board.

I tried to flash the WCHLINK_V1.2.bin via wchisptool and it worked! However at first only one red led came up and no hid or uart devices were visible on usb and vid/pid was 1a86/8010 but I connected TX to GND as per readme there and it helped, since then pid is 8011 and both hid and uart comes up. Tested with nrf52 board and both uart and SWD debugging works :-)

Interesting is that previously VID/PID was **2**a86/8011 and bcdDevice 1.00 and now it is  **1**a86/8011 and bcdDevice is 1.02. So I had something different there but it still works and I can confirm this is indeed CH549 chip and flashing works :-)