DeqingSun/ch55xduino #61: Upload fails on MacOS 11.6 (M1 Mac)

akita11, 2021-12-19:
I'm trying to upload a sketch to CH552 board (in DFU mode) on MacOS 11.6 (M1 Mac), and results in the following error. 
--------
dyld: Library not loaded: /usr/local/opt/libusb/lib/libusb-1.0.0.dylib
  Referenced from: /Users/akita/Library/Arduino15/packages/CH55xDuino/tools/MCS51Tools/2020.12.01/macosx/vnproch55x
  Reason: image not found
--------
Separately I've installed libusb by homebrew, and the directry strucute seems to differ from my environemnt, where libusb is istalled in /opt/homebrew/lib.

DeqingSun, 2021-12-19:
The lib was statically linked to avoid issue for novice users. And at that time they don't have ARM mac.

Try to compile this one:
https://github.com/DeqingSun/vnproch551

You may need to change the makefile to link the right lib, or you just change it to use dynamic link.

akita11, 2021-12-19:
Thanks, I'll try it, and if it is OK, I'll close this issue.

akita11, 2021-12-19:
I've tried to compile vnproch551 on Mac(M1), with some modifications as below, with libusb and libusb-compat installed by brew, and replaced vnproch551 with original one (in /Users/(Username)/Library/Arduino15/packages/CH55xDuino/tools/MCS51Tools/2020.12.01/macosx). I've succeeded to dowonload!

Here is my modifications on Makefile. (add include path, renamed libusb's directory name and version numbers)
```
	LIBS=/opt/homebrew/Cellar/libusb/1.0.24/lib/libusb-1.0.a /opt/homebrew/Cellar/libusb-compat/0.1.5_1/lib/libusb.a -ldl -Wl,-framework,IOKit -Wl,-framework,CoreFoundation
	INCLUDE=-I /opt/homebrew/Cellar/libusb/1.0.24/include
...
%.o:%.cpp
	$(CPP) $(INCLUDE) -c $< -o $@
```

DeqingSun, 2022-03-21:
@akita11 Hi akita11, I updated the vnproch551 tool to support the CH375 driver in Windows, so windows user can use CH375 (official), libusb or winusb with the bootloader. Hopefully it eliminate the hassle of changing driver.

Meanwhile, I'd like to update the tool in a new release of CH55xduino. I'm wondering, would you do me a favor and compile a universal binary of vnproch551?I don't have access to a m1 mac now.

akita11, 2022-03-21:
OK, I'll build it soon, and tell you.

akita11, 2022-03-22:
@DeqingSun I've tried to build vnproch551 on my M1 mac, with modification on lib&include pathes for my Homebrew environment, and it results the error at link:
```
Undefined symbols for architecture arm64:
  "_SecTaskCopyValueForEntitlement", referenced from:
      _darwin_detach_kernel_driver in libusb-1.0.a(darwin_usb.o)
  "_SecTaskCreateFromSelf", referenced from:
      _darwin_detach_kernel_driver in libusb-1.0.a(darwin_usb.o)
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [vnproch55x] Error 1
```
I found its solution [here](https://githubhot.com/repo/libusb/libusb/issues/1072), and succeeded to build it.
Although I didn't checked it yet, I've placed in my forked repo as vnproch551_MacM1:
https://github.com/akita11/vnproch551

akita11, 2022-03-22:
> @akita11 <https://github.com/akita11> Thanks for the update,
>
> I downloaded this file, but it still said
>
> -bash: ./vnproch55x_MacM1: Bad CPU type in executable
>
> I'm using a 2014 Macbook Pro with 10.14.6

It seems to be a M1-specified binary, I will survey and try how to build
universal binary.

DeqingSun, 2022-03-23:
@akita11 thank you for the help. By the way someone reported there is a new 2.5.0 version bootloader appears and there will be an update for the tool soon.
https://github.com/DeqingSun/ch55xduino/issues/65

However, in my experience with ch573, that chip reports 3.6.0 version of bootloader but still with uses the same protocol.

DeqingSun, 2022-03-24:
@akita11 
Hi akita11,
Just let you know, please pull the new updates to accept the 2.5.0 bootloader before build the tool. There is new bootloader on the market.

akita11, 2022-03-25:
@DeqingSun I've learned about universal binary on Mac, and we can use Intel binary on M1 Mac by automatic emulation, Rosetta2. So, I think we don't have to prepare "universal binary for Intel and M1". Actually, some of my firends succeeded to execute vnproch55x (for Intel Mac) on M1 Mac with no special procedure, like replacing binary.

I think the reason why I fails to execute your vnproch55x on my M1 Mac is the mismatch of the path for libusb.
I get the errror message when executing vnproch55x on my M1 Mac:
```
% ./vnproch55x 
dyld: Library not loaded: /usr/local/opt/libusb/lib/libusb-1.0.0.dylib
  Referenced from: /Users/akita/Desktop/./vnproch55x
  Reason: image not found
zsh: abort      ./vnproch55x```

I'm using [homebrew (3.0.0)](https://brew.sh/2021/02/05/homebrew-3.0.0/), which is installed in /opt/homebrew. For example. libusb-1.0.0.dylib is installed at /opt/homebrew/lib/libusb-1.0.0.dylib.

I'm continueing to how to handle this matter, but do you have any idea for solution?

DeqingSun, 2022-03-25:
@akita11 Thanks for the feedback, I'll see if static link will be a solution.

DeqingSun, 2022-03-25:
@akita11 
I think it is a static link issue. I tried to rename the /usr/local/opt/libusb/lib/libusb-1.0.0.dylib in my computer, a fresh built tool worked while the one come with ch55xduino 0.0.9 did not work.

Then I realized the tool in 0.0.9 was compiled before the [static link change](https://github.com/DeqingSun/vnproch551/commit/081b30dbaa402f48892f8db10ed4479fd615aa17). So please try if the fresh compiled one works for you or not.

[vnproch55x.zip](https://github.com/DeqingSun/ch55xduino/files/8347051/vnproch55x.zip)

akita11, 2022-03-25:
I've tried your binary on my M1 Mac, and OK to flash !

closed 2022-03-25 by DeqingSun