DeqingSun/ch55xduino #63: How to enter Bootloader for CH552 from code?

lpuser13, 2022-02-28:
Hi
 I have been using this core for CH552 successfully for a while now....tnx

I am now trying to activate the Bootloader (built-in default v02.31) from within the code - without success.
_(I can enter the booloader using DP+ approach using WCHISPTool{v2.70} to download firmware_)

I have tried LJMP 0x3800, but it never seems to enter the bootloader.

Does anyone have a working example available?

_FYI: my config in the IDE is USB with 148B USB RAM @ 24MHz_

lpuser13, 2022-02-28:
Worked fine, many thanks for the support and prompt response.

closed 2022-02-28 by lpuser13

perkinsb1024, 2025-01-01:
> @lpuser13 check
> 
> https://github.com/DeqingSun/ch55xduino/blob/78d01729a897dd2080c855575f0042dc18c31fc4/ch55xduino/ch55x/cores/ch55xduino/USBCDC.c#L49

Just in case anyone else comes across this and it doesn't work for them, I found that a [newer commit](https://github.com/DeqingSun/ch55xduino/blob/ch55xduino/ch55xduino/ch55x/cores/ch55xduino/USBCDC.c#L69) adds the line `TMOD = 0`, which was required for me to enter bootloader mode, otherwise the `lcall #0x3800` command would not enter bootloader