DeqingSun/ch55xduino #65: support for bootloader 2.5.0

sgmklp, 2022-03-22:
I use the CH552 to try it, but get this
[image]
Then I see the JavaScrip Code, find it seems not support bootloader 2.5.0

DeqingSun, 2022-03-22:
I'll compile a new windows version later today.

DeqingSun, 2022-03-22:
@sgmklp please try this tool. You can replace the original one in

```
C:\Users\YOUR USER NAME\AppData\Local\Arduino15\packages\CH55xDuino\tools\MCS51Tools
```

[vnproch55x.exe.zip](https://github.com/DeqingSun/ch55xduino/files/8327159/vnproch55x.exe.zip)

This file used the same protocol but accepts 2.5.0 as bootloader version.

DeqingSun, 2022-03-24:
@sgmklp Does the updated version work with you?

closed 2022-03-24 by DeqingSun

DeqingSun, 2022-03-31:
There was a logic bug that made the tool only support 2.5.0 version of bootloader. 

This one is better.
[vnproch551_exe_20220331.zip](https://github.com/DeqingSun/ch55xduino/files/8390855/vnproch551_exe_20220331.zip)

sgmklp, 2022-03-31:
OH, thanks.