DeqingSun/ch55xduino #69: Change USB DM ( p3.7) pinMode problem

Kaitoukid93, 2022-05-10:
I have the following code in setup just for detecting the type of host device. The detection is working wonderful!
When I tried to set the pinMode of P3.7 (USB DM) to Output and drive it low, it did not change the line state ( the line is pulled high by 10k resistor in the host side)
Is anything i'm doing wrong or the pinMode can not change when using USB Library?
Thanks
```
void setup()
{

  pinMode(36, INPUT);
  pinMode(37, INPUT);
  bool DMState;
  DMState = digitalRead(37);
  if (DMState == HIGH)
    Host = PowerLED;
  else
    Host = Computer;
  switch (Host)
  {
    case Computer: // this USB is attached to Host Computer

      USBInit();

      }

      break;

    case  PowerLED : // this usb is attached to the  Customized power brick

     
      pinMode(37, OUTPUT);
      digitalWrite( 37, LOW); //drive DM LOW to notify the power brick
      
      break;
  }
```

DeqingSun, 2022-05-10:
When USB is enabled, bUSB_IO_EN is set to 1, and you can not make the pin to output. The datasheet mentioned this.

DeqingSun, 2022-05-10:
If you are sure you are not going to use USB anymore, try

PIN_FUNC &= ~bUSB_IO_EN;

Before you use the gpio again.

Kaitoukid93, 2022-05-10:
> If you are sure you are not going to use USB anymore, try
> 
> 
> 
> PIN_FUNC &= ~bUSB_IO_EN;
> 
> 
> 
> Before you use the gpio again.

Thanks! That's exactly what I need. When the host is not the computer,I don't need usb anymore

closed 2022-05-11 by Kaitoukid93