DeqingSun/ch55xduino #72: CMSIS_DAP example build error

reppad, 2022-06-08:
Hi,

I tried to compile the CMSIS_DAP example with ch55xduino 0.0.12 but compilation fail when trying to compile "SW_DP.c" file :

```
[...]
/home/reppad/.arduino15/packages/CH55xDuino/tools/MCS51Tools/2022.05.06/wrapper/sdcc.sh /home/reppad/.arduino15/packages/CH55xDuino/tools/sdcc/build.13407/bin/sdcc /tmp/arduino_build_281224/sketch/src/CMSIS_DAPusb/SW_DP.c /tmp/arduino_build_281224/sketch/src/CMSIS_DAPusb/SW_DP.c.o re1 -MMD -c -Ddouble=float -DUSE_STDINT -D__PROG_TYPES_COMPAT__ --less-pedantic -mmcs51 -DCH551 -DF_CPU=24000000L -DARDUINO=10815 -DARDUINO_ch55x -DARDUINO_ARCH_MCS51 --model-small -DUSER_USB_RAM=148 -I/home/reppad/.arduino15/packages/CH55xDuino/hardware/mcs51/0.0.12/cores/ch55xduino -I/home/reppad/.arduino15/packages/CH55xDuino/hardware/mcs51/0.0.12/variants/ch552 -I/home/reppad/.arduino15/packages/CH55xDuino/tools/sdcc/build.13407/share/sdcc/include
Caught signal 11: SIGSEGV
cp: impossible d'évaluer '/tmp/arduino_build_281224/sketch/src/CMSIS_DAPusb/DAP.c.o': Aucun fichier ou dossier de ce type
exit status 1
Error compiling for board CH551 Board.
```

Same behavior with the 4 supported boards.
Same behavior on Linux and Windows.

I tried to downgrade few version to test if it is a regression:
0.0.12 : KO
0.0.11 : KO
0.0.10 : OK
0.0.9  : OK

So it seems this example does not build from 0.0.11 release.

Thanks !

DeqingSun, 2022-06-09:
@reppad This is a compiler bug.

https://sourceforge.net/p/sdcc/bugs/3401/

You can either use a old version, or change num to uint32_t type.

reppad, 2022-06-09:
Thanks @DeqingSun,
Indeed it works by changing the type of num.
Do you want me to do a PR with this workaround or do you prefer to wait for a fixed version of SDCC?

DeqingSun, 2022-06-09:
@reppad 
I'll wait for a fixed version of SDCC

DeqingSun, 2022-08-12:
Bug seems fixed. I will check again later when I do the next release

DeqingSun, 2022-09-18:
Although this bug is fixed, the current SDCC build introduces some new worse bugs. I'll wait till there is a fix.

goddade, 2022-12-22:
It still doesn't compile in version 0.0.16.
···
Caught signal 11: SIGSEGV
cp:: No such file or directory

exit status 1

Compilation error: exit status 1
···

DeqingSun, 2023-05-10:
SDCC seems slow on fixing bugs, use the workaround for now.

closed 2023-05-10 by DeqingSun