DeqingSun/ch55xduino #76: Upload with CDC fails on Arduino 2.0-rc7

kaofishy, 2022-06-23:
When using the latest Arduino 2.0 release candidate, upload on a CH552 already programmed with a CDC enabled sketch fails with the following error:
```
------------------------------------------------------------------
CH55x Programmer by VNPro
------------------------------------------------------------------
Load file as hex
   Loaded 4047 bytes between: 0000 to 101A
Found no CH55x USB
```
I was able to replicate this on both Linux and Windows with multiple boards. Manually triggering the USB bootloader and then upload works however.

This issue does not happen on Arduino 1.8.

I suspect there's a timing issue where there's not enough time to enumerate the device between the 1200bps trigger and the programming step, since the whole upload process feels much faster on 2.0 compared to 1.8.

DeqingSun, 2022-06-23:
@kaofishy Thanks for reporting. The Arduino 2.0-rc7 works well on a Mac computer, and I think the Mac recognize USB much faster than other systems. I'll setup a Windows computer later to test.

DeqingSun, 2022-06-25:
@kaofishy The bug can be reproduced on a Win10 desktop computer with Arduino 2.0-rc7, I'll try to make the upload code try for multiple times.

DeqingSun, 2022-06-26:
@kaofishy
Please replace the platform.txt and vnproch55x.exe. And then see if it works. 

[platform.txt](https://github.com/DeqingSun/ch55xduino/files/8985802/platform.txt)
[vnproch55x.zip](https://github.com/DeqingSun/ch55xduino/files/8985803/vnproch55x.zip)

On my computer the upload starts about 300ms after the execution vnproch55x.exe.

kaofishy, 2022-06-26:
@DeqingSun I think you accidentally attached the macOS binary instead of the Linux and Windows ones.

kaofishy, 2022-06-26:
@DeqingSun FYI I compiled it from source from your repo and it does work on my Linux machine now :+1:

DeqingSun, 2022-06-26:
@kaofishy You are right I attached the Mac version.
Try this one.
[vnproch55x.exe.zip](https://github.com/DeqingSun/ch55xduino/files/8986008/vnproch55x.exe.zip)

kaofishy, 2022-06-26:
@DeqingSun Just tested it on Windows, it's working consistently now. FYI I can confirm that on my machine it takes about 0.5s (~four retries) to connect to the bootloader. Thanks for looking into this so promptly.

closed 2022-09-18 by DeqingSun

domdfcoding, 2022-11-09:
On the 2.0 IDE I can't find an option for uploading with CDC, only via serial (and the serial port doesn't show). Works fine on the old IDE. But after I uploaded through the old IDE with CDC the serial port shows up with the 2.0 IDE and I can upload through there.

Strange, and I think it's probably the new IDE being wonky, but I thought I'd mention it in case someone else runs into the same problem.

Thanks for the library!

DeqingSun, 2022-11-10:
@domdfcoding 
I'm not sure if you mean you can not see the "USB" upload option. I've tested 2.0.1 on Mac and 2.0.0 IDE on Win10 without CH552 plugged. And they can start the uploading process.
[image]
[image]

The upload process is not done with the CDC but with the bootloader code. If you have CDC, the upload process can send signal to CH552 and let it jump into bootloader (no longer CDC), or you can enter the bootloader manually by holding a button and plug in the board. Either way the uploading program will communicate with the bootloader, not the CDC interface.