DeqingSun/ch55xduino #78: src文件夹内包含userUsbHidKeyboard和userUsbCdc的文件夹时刷什么都会识别成keyboard

15010200730, 2022-06-28:
src文件夹内包含userUsbHidKeyboard和userUsbCdc的文件夹时上传什么都会识别成keyboard

DeqingSun, 2022-06-28:
userUsbHidKeyboard can not coexist with userUsbCdc.

You need to merge the code if you want to use both.

15010200730, 2022-06-29:
> 

Thanks for your reply! My English is not good, sorry!
I mean, I want to use "`userusbcdc`" to communicate with the host computer to change the keyboard key value
I want to use an IO of ch552 to control the code entering `"userusbcdc"` or `"userusbhidkeyboard"`
I tried to add the loop in the two files in the example to while (1) and then convert it into a function and put it in the setup function. Only after "/src" was placed in the folder of "userusbcdc", the code of running userusbcdc was normal, but after I put the file of userusbhidkeyboard into the file, the same code was brushed into ch552 and the system recognized it as `"hidkeyboard"`, or vice versa
Could you tell me what I did wrong? What should I do? I'm not very familiar with Arduino
This problem has been bothering me for a week.......TxT....
If you can help me, I'd like to thank you very much
Think ！^_^

DeqingSun, 2022-06-29:
Both userusbcdc and userusbhidkeyboard contains USBhandler.c and USBconstant.c, and that create a conflict. 

You will need to merge those files to one, and adjust usb descriptors and handlers. That does not require any Arduino specific knowledge, but it requires knowledge of USB communication.

15010200730, 2022-06-29:
> Both userusbcdc and userusbhidkeyboard contains USBhandler.c and USBconstant.c, and that create a conflict.
> 
> You will need to merge those files to one, and adjust usb descriptors and handlers. That does not require any Arduino specific knowledge, but it requires knowledge of USB communication.

原来是这样..谢谢！我刚刚接触电子不就，hid协议也没有认真的看过，我会再研究一下（尽管大概率不行），实在不行我再想一些其他的办法解决吧！最后真的感谢您！

15010200730, 2022-06-29:
> Both userusbcdc and userusbhidkeyboard contains USBhandler.c and USBconstant.c, and that create a conflict.
> 
> You will need to merge those files to one, and adjust usb descriptors and handlers. That does not require any Arduino specific knowledge, but it requires knowledge of USB communication.

原来是这样..谢谢！我刚刚接触电子不久，hid协议也没有认真的看过，我会再研究一下（尽管大概率不行），实在不行我再想一些其他的办法解决吧！最后真的感谢您！

15010200730, 2022-06-30:
> Both userusbcdc and userusbhidkeyboard contains USBhandler.c and USBconstant.c, and that create a conflict.
> 
> You will need to merge those files to one, and adjust usb descriptors and handlers. That does not require any Arduino specific knowledge, but it requires knowledge of USB communication.

After a day's hard work, I still haven't solved this problem ..
Thank you very much for your reply

closed 2022-08-12 by DeqingSun