DeqingSun/ch55xduino #82: pulseIn is error，HOW TO USE HC-SR04?超声波测距提示pulseIn错误

wukongxuetang, 2022-07-23:
I want to get distance with  a HC-SR04 using code:
  digitalWrite(trig, HIGH);
  delayMicroseconds(10);
  digitalWrite(trig, LOW);
  distance  = pulseIn(echo, HIGH,100000);
  distance  = distance / 58;
but have a wrong about "pulseIn",it is "function 'pulseIn' implicit declaration";when add code：
  extern unsigned long pulseIn(uint8_t pin, uint8_t state, unsigned long timeout);
have other error：
  
then I found the pulseIn has been commented-out in CH55xDuino\hardware\mcs51\0.0.12\cores\ch55xduino\Arduino.h
[image]

DeqingSun, 2022-07-23:
That is not implemented yet.

wukongxuetang, 2022-07-23:
hope "pulseIn" can be used as early as possible,thanks

DeqingSun, 2022-08-12:
It is difficult to implement a universal PulseIn on 8051. Unlike AVR, you can not access a port register from a pointer.  

Here I made some functions you can use:

```
void setup() {
  pinMode(14, INPUT);
}

unsigned long pulseIn14WithTimeout() {
  uint32_t pulseInTimeOut = 1000000UL;
  uint32_t pulseInStartTime = micros();
  uint32_t pulseInRisingTime, pulseInFallingTime;

  while (P1_4 == 0) {
    if (((int32_t)(micros() - pulseInStartTime)) > pulseInTimeOut) {
      return 0;
    }
  }
  pulseInRisingTime = micros();
  while (P1_4 == 1) {
    if (((int32_t)(micros() - pulseInStartTime)) > pulseInTimeOut) {
      return 0;
    }
  }
  return (micros() - pulseInRisingTime);
}

unsigned long pulseIn14() {
  uint32_t pulseInRisingTime;

  while (P1_4 == 0) {
  }
  pulseInRisingTime = micros();
  while (P1_4 == 1) {
  }
  return (micros() - pulseInRisingTime);
}

void loop() {
  USBSerial_println(pulseIn14());
  delay(1000);
}
```

The function uses a fixed pin, and that can run fast. If you need higher resolution, you can use the ```pulseIn14``` function to trade timeout protection for higher resolution. Or you just use the safer ```pulseIn14WithTimeout```.

You can assign other pins or change logic by changing ```P1_4 == 0``` and ```P1_4 == 1```.

Nable80, 2022-08-12:
Btw, you don't need these casts to ``int32_t``. It's wrong (unsigned subtraction already handles overflow properly, e.g. ``1 - UINT_MAX == 2``) and effectively useless (comparison between signed and unsigned integers implicitly casts signed back to unsigned).

DeqingSun, 2022-08-12:
@Nable80 The disassembly result proves that you are right. I thought that was my habit when using it with int16_t to make it slightly faster in some cases.

DeqingSun, 2022-08-12:
And this pulseIN seems safer

```
void setup() {
  pinMode(14, INPUT);
}

unsigned long pulseIn14WithTimeout() {
  uint32_t pulseInStartTime = micros();
  uint32_t pulseInRisingTime;

  while (P1_4 == 0) {
    if (((int32_t)(micros() - pulseInStartTime)) > 1000000L) {
      return 0;
    }
  }
  pulseInRisingTime = micros();
  while (P1_4 == 1) {
    if (((int32_t)(micros() - pulseInStartTime)) > 1000000L) {
      return 0;
    }
  }
  return (micros() - pulseInRisingTime);
}

unsigned long pulseIn14() {
  uint32_t pulseInRisingTime;

  while (P1_4 == 0) {
  }
  pulseInRisingTime = micros();
  while (P1_4 == 1) {
  }
  return (micros() - pulseInRisingTime);
}

void loop() {
  USBSerial_println(pulseIn14WithTimeout());
  delay(1000);
}
```

Nable80, 2022-08-12:
This is an infamous part of the C standard and it's easy to come across it if you use gcc/clang with ``-Wall -Wextra … -Weverything`` daily. It's sad that SDCC can't provide this level of static analysis but it's still much better than nothing.

Unfortunately these standards have a lot of obscure parts and sometimes they bite painfully. Assembler listings are really useful but they're not enough when you come across the so-called implementation-defined or (more often) undefined behavior (UB) and both C and C++ are full of it.

Implicit cast to unsigned is quite harmless and even useful, while the other type of (completely non-obvious) implicit casts often drives me mad: it occurs when one of the operands is smaller than ``int``, then both operands are cast to *signed* ``int`` even if both of them were unsigned. Sometimes it may be useful (e.g. while dealing with chars) but most often it's awful and makes me add tons of explicit casts to ``unsigned`` to keep the compiler happy. No, I can't just disable ``-Wsign-conversion`` warnings, sometimes they highlight real bugs, so it's better to fix such places instead of hiding them.

---
Speaking about pointers to I/O registers, they can be implemented with switch/case (to read the whole P0…P4 depending on the selected pin) and bit-masks (to check the corresponding bit in a fetched byte value). We don't have many I/O ports, so it should be possible to implement it this way without much bloat. But I don't think it's a good idea to poll pins for time measurement anyway, MCU-specific timers should be used instead.

DeqingSun, 2022-08-12:
Thanks for sharing the implicit cast things. I recalled the tons of "warning comparison between signed and unsigned integer" when I was compiling MicroPython (or something similar).

------

Yes from an engineer's perspective, the leftover timer1 should be used to do gated timing. Although we still need to poll the overflow often to deal with long pulse. 

But when I see the pulseIn as a teacher, it seems quite beautiful. There is no restriction on pin number and there is timeout protection. And there is just 1 line of code. I actually thought about how to do it on 8051. The pulling might be done with an array of JB/JNB/MOV, and use ```JMP @A+DPTR``` to choose the right instruction.

DeqingSun, 2023-10-04:
@wukongxuetang Now there is pulseIn

@Nable80 I found a look up table is ideal for MCS51 to read arbitrary pin fast. __sdcc_call_dptr is quite handy to call function at any address. The lookup table just move the pin bit to carry and return. overall it takes about 10~20 clk to read a pin.

Nable80, 2023-10-06:
@DeqingSun thank you for a tip about ``__sdcc_call_dptr``!
I think the lookup table can still be compacted, although arduino's abuse of decimal numbers for pins makes it somehow complicated, at least if we care about the latency… Maybe ``DA`` (decimal adjust accumulator) instruction could help a bit but I didn't think enough about it.

closed 2023-11-01 by DeqingSun