DeqingSun/ch55xduino #83: CH559T schematic

ITstreet1, 2022-08-11:
Not sure if this is a good place...

I would like to make a board based on CH559T. There is very little data about the design of it. All I have found so far is:
https://ae01.alicdn.com/kf/H75820655d97742698b2b93e22eca0233w.jpg

I have a few questions:

1. According to the schematic, MCU is powered by 3.3V. What is the purpose of the VIIN5 pin? Also those three resistors under the MCU itself?
2. There are 3 USB ports. All connected at the same pins. This MCU should have a USB and a USB Host. Some light here.
3. What is the purpose of the UART (Download) header? There is a button for this purpose, too.
4. I believe using this (any of) USB, and JSON from here, I could program it with Arduino IDE.

These are just my thoughts. I believe this could be useful to others, too.

Thank you in advance.

ITstreet1, 2022-08-11:
It's this board from the schematic
https://www.aliexpress.com/item/1005002726995714.html

DeqingSun, 2022-08-12:
It depends on whether you want 5V or 3.3V for IO. If 5V, you don't need external LDO and the on chip LDO will do the job. Those resistors in schematic are jumpers for configuration. You shall check the data sheet for circuit design.

Also here is another design you can refer to.
https://github.com/akita11/CH559Pico/blob/main/CH559Pico_sch.pdf

This MCU should have a USB **OR** a USB Host. You can only choose one. The pins are the same. 

The description of the board is horrible. I have such a board. And I know the original designer of this board knows the chip well. The UART is just UART, you can use it to flash firmware instead of USB. 

The button is to kick the board into boot loader. You hold the button when you power on the board, and then the device will enter bootloader.

And yes you can program it with Arduino IDE.

ITstreet1, 2022-08-12:
If I get you correctly, I can use the schematic from here for CH552e and it would work.

DeqingSun, 2022-08-12:
I have no idea why you are talking about CH552e

ITstreet1, 2022-08-12:
The basic schematic here for the CH552E dev board is almost identical to the CH559T. Almost to copy it and paste it here.

DeqingSun, 2022-08-13:
The schematic in this repo is using CH552G, not CH552E.

You also need to move the bootloader button to the correct pin on CH559.

ITstreet1, 2022-08-14:
My bad. Sorry.
552G and 552E are basically the same. I have a bunch of CH55x...
About the bootloader button, I think it is the only difference.

Thank you for your info. It really helped.

closed 2022-08-14 by DeqingSun