DeqingSun/ch55xduino #87: Enable touch pad caused system failure

kahoch, 2022-09-02:
I'm using a CH552T which powered by 3.3V LDO, if use "TouchKey_begin( )" will cause system failure, no metter how many channels. Once I comment it,  the system works fine. And the same code works flawless on a 5V system. Is there any config I missed?

kahoch, 2022-09-03:
After consulting WCH, a possible cause is frequency. Maybe 16M is still too high, how can I make it run on 12Mhz?

DeqingSun, 2022-09-03:
Please refer to wiring.c for frequency change. If you are doing CH552, you should do something like this before you initialize the touch key module.

```
    //set internal clock 
    SAFE_MOD = 0x55;
    SAFE_MOD = 0xAA;
    
    CLOCK_CFG = CLOCK_CFG & ~ MASK_SYS_CK_SEL | 0x04;  // 12MHz

    SAFE_MOD = 0x00;

    delayMicroseconds(5000); //needed to stablize internal RC
```

Please note doing so will cause the all timing got inaccurate. Including millis micros delay and hardware serial. The USB should be fine. If you experiment proves that the 12Mhz can solve the problem, I'll add the 12Mhz support with additional code to solve the timing issues.

kahoch, 2022-09-04:
> 

It works like a charm! And USB serial works too. 
Thank you for your great work!

BTW, according to datasheet, touch input should be fine on 16Mhz on 3.3V, but it really doesn't. I've asked WCH to verify this issue, any updates I'll post here.

DeqingSun, 2022-09-18:
12Mhz option is included in the new 0.13 release

closed 2022-09-18 by DeqingSun

kahoch, 2022-09-20:
After several emails, I still cannot get any useful info from WCH.