DeqingSun/ch55xduino #88: SPI in ch55xduino help

jnthas, 2022-09-05:
Hello! I'm writing this issue with hope that you could give me some help. I'm facing a strange behavior with CH552 dev board and the module W25Q32 (external flash memory) when I'm reading data when I using the SPI and ch55xduino. Let me explain step-by-step what happens, and any help is appreciated:

- I've ported a library to read SPI flash chips, merged here https://github.com/DeqingSun/ch55xduino/tree/ch55xduino/ch55xduino/ch55x/libraries/SPIFlash
- I have the W25Q32 all erased
- I write the same byte (e.g. 0xAA) in all of the 16 pages of the first sector (total of 4096 bytes. Each page contains 256 bytes)
- When I read the sector, I have inconsistent data, some bytes are OK but other not (example below)

What is making me think the issue is related with ch55xduino:
- If I read using a regular Arduino, it works, the whole sector (4096) are fulfilled with the byte I have written (0xAA).
- After some tries, I also could read the flash correctly using this project as start point https://github.com/WeActStudio/WeActStudio.CH552CoreBoard/tree/master/SDK/03-SDCCExample/01-Blink 

Here is the example of readings, notice there is a pattern in the readings, only the page 1,2,4,8 and 16 contains the byte I wrote. 
```
//I'm reading the flash like this:
for (int j=0; j<16; j++) {    
  readBytes((j*256), _sbuffer, 255);
    
  for (uint8_t i=0; i<255; i++) {
    USBSerial_print(_sbuffer[i], HEX);
    USBSerial_print(".");    
  }  
}

OUTPUT:

Reading addr: 0 until 255
1) AA.AA.AA.AA.AA.AA.AA.AA.AA.AA. . (abbreviated for brevity reason) . .AA.AA.AA.AA.

Reading addr: 256 until 511
2) AA.AA.AA.AA.AA.AA.AA.AA.AA.AA. . . .AA.AA.AA.AA.

Reading addr: 512 until 767
3) FF.FF.FF.FF.FF.FF.FF.FF.FF.FF. . . .FF.FF.FF.FF.

Reading addr: 768 until 1023
4) AA.AA.AA.AA.AA.AA.AA.AA.AA.AA. . . .AA.AA.AA.AA.

Reading addr: 1024 until 1279
5) FF.FF.FF.FF.FF.FF.FF.FF.FF.FF. . . .FF.FF.FF.FF.

Reading addr: 1280 until 1535
6) FF.FF.FF.FF.FF.FF.FF.FF.FF.FF. . . .FF.FF.FF.FF.

Reading addr: 1536 until 1791
7) FF.FF.FF.FF.FF.FF.FF.FF.FF.FF. . . .FF.FF.FF.FF.

Reading addr: 1792 until 2047
8) AA.AA.AA.AA.AA.AA.AA.AA.AA.AA. . . .AA.AA.AA.AA.

Reading addr: 2048 until 2303
9) FF.FF.FF.FF.FF.FF.FF.FF.FF.FF. . . .FF.FF.FF.FF.

Reading addr: 2304 until 2559
10) FF.FF.FF.FF.FF.FF.FF.FF.FF.FF. . . .FF.FF.FF.FF.

Reading addr: 2560 until 2815
11) FF.FF.FF.FF.FF.FF.FF.FF.FF.FF. . . .FF.FF.FF.FF.

Reading addr: 2816 until 3071
12) FF.FF.FF.FF.FF.FF.FF.FF.FF.FF. . . .FF.FF.FF.FF.

Reading addr: 3072 until 3327
13) FF.FF.FF.FF.FF.FF.FF.FF.FF.FF. . . .FF.FF.FF.FF.

Reading addr: 3328 until 3583
14) FF.FF.FF.FF.FF.FF.FF.FF.FF.FF. . . .FF.FF.FF.FF.

Reading addr: 3584 until 3839
15) FF.FF.FF.FF.FF.FF.FF.FF.FF.FF. . . .FF.FF.FF.FF.

Reading addr: 3840 until 4095
16) AA.AA.AA.AA.AA.AA.AA.AA.AA.AA. . . .AA.AA.AA.AA.

```

DeqingSun, 2022-09-18:
You'd better get a logic analyzer to see where the problem really is.

jnthas, 2022-09-19:
Yes, I just bought one, waiting it arrives. ch55xduino make things a lot easier.. it needs to work.

DeqingSun, 2022-11-01:
@jnthas Did you figure out the reason? Can you provide the signal capture of the wrong reading and right reading?

jnthas, 2022-11-02:
Hey! It turns out I keep developing using SDCC instead of Arduino, I'm almost finishing the project. I will record the signal to send here soon.

DeqingSun, 2023-03-13:
@jnthas Are you planning to get the bug troubleshooted? If not the issue will be closed and your contribution will be reverted.

jnthas, 2023-03-13:
I'll close the issue for now. I didn't have much time to investigate the cause. Let me know if someone may have the same issue so we can work together.

closed 2023-03-13 by jnthas