DeqingSun/ch55xduino #89: Enter and Exit low power modes examples

danilomna, 2022-09-07:
Hi,

Would be possible add low power examples to the lib?
This kind of feature is essential to battery powered devices and would be very usefull.

Enter Low power, sleep mode, deep sleep
Exit Low power, IO level change, timer interrupt, watchdog

Regards

Danilo

DeqingSun, 2022-09-24:
Read the datasheet before you ask questions like that.

This chip does not have that many sleep features

closed 2022-09-24 by DeqingSun

danilomna, 2022-09-26:
Read the datasheet before answer a question like that.

The chip have sleep feature.

[image]

danilomna, 2022-09-26:
Please reopen issue.

guibom, 2022-10-23:
I'm curious about this too... I checked the datasheet and can see some defined registers on ch5xx.h, to enable specific wake up events. 
But I'm not sure how to use those.