DeqingSun/ch55xduino #90: WS2812 without delay

kbln, 2022-09-16:
Hi all! 

```
neopixel_show_ P1_5(ledData, NUM_BYTES); // does not work without delay(1)
delay(1);
```

How can i make it work without delay?

DeqingSun, 2022-09-16:
The delay has to be there. But you may change it to delayMicroseconds(50). Because you need to give the WS2812 at least 50 us to reset its data. If you don't have other code takes longer than 50us, use a delay or delayMicroseconds.

kbln, 2022-09-16:
> The delay has to be there. But you may change it to delayMicroseconds(50). Because you need to give the WS2812 at least 50 us to reset its data. If you don't have other code takes longer than 50us, use a delay or delayMicroseconds.

Thanks for the answer! I will try delayMicroseconds(50)

I am trying to run [Adalight](https://github.com/Wifsimster/adalight_ws2812) on ch552G, but Serial is very slow. I think the delay is the problem.

closed 2022-12-02 by DeqingSun