DeqingSun/ch55xduino #91: 0.014版本ADC和外部晶振问题

perseverance51, 2022-09-30:
非常感谢您对本项目的持续维护！在使用当前版本0.014版本测试当中，发现以下2点bug：
1. 选择型号CH552，跑AnalogReadSerial示例，读取的ADC值比较小，范围：0-127，相同示例下，选择CH559，读取ADC值范围：0-2047；
 2. CH559选择外部24MHz晶振时，编译正常，运行时无法识别到CDC虚拟串口，选择内部24MHz正常，有CDC虚拟串口。
V0.014解决了CDC虚拟串口报错问题，但是并没有实现通过CDC虚拟串口实现代码的上传，还是需要借助WCHISPTool工具进行程序的烧录。其他示例和功能还未进行测试。

DeqingSun, 2022-09-30:
@perseverance51 

选择型号CH552，跑AnalogReadSerial示例，读取的ADC值比较小，范围：0-127
I got 255 when I test with CH552 and connect P1.1 to 5V.
The 0~2047 on CH559 is intentional, wiring_analog.c samples analog pin with 11bits.

CH559选择外部24MHz晶振时，编译正常，运行时无法识别到CDC虚拟串口.
The external oscillator option assumes your external oscillator works at the same frequency as the internal one, which is 12MHz. I've tested my CH559 with 12M external oscillator and they worked. If you have another frequency, you need to set CLOCK_CFG.

The CDC upload has been tested on Mac OS, Win10 and Win XP virtual machine. They all can upload with CDC.
[image]
[image]

If you still get errors, would you provide your error message?

perseverance51, 2022-10-07:
- 下面是上传失败的信息
`------------------------------------------------------------------
CH55x Programmer by VNPro
------------------------------------------------------------------
Load file as hex
   Loaded 8571 bytes between: 0000 to 21C8
ch375Version 35
CH375GetUsbID 55e04348
CH375 open OK
MCU ID: 59 11
Found Device CH559
Bootloader: 2.4.0
ID: A2 FD 4B BC
XOR Mask: A6 A6 A6 A6 A6 A6 A6 FF 
Write 8649 bytes from bin file.
..................................................[2K
Write complete!!!
Verify chip
......................................上传项目出错
...............................................[2K
Packet 146 doesn't match.
`
-
- 似乎和编译产生的Hex文件大小有关，所编译的工程项目越大，上传失败的风险越高。下面是成功上传信息

`------------------------------------------------------------------
CH55x Programmer by VNPro
------------------------------------------------------------------
Load file as hex
   Loaded 4776 bytes between: 0000 to 12F4
ch375Version 35
CH375GetUsbID 55e04348
CH375 open OK
MCU ID: 59 11
Found Device CH559
`
- 上传失败每次都是卡在执行到同一个地方报错：
`..................................................[2K
Write complete!!!
Verify chip
......................................上传项目出错
...............................................[2K
Packet 146 doesn't match.`
- 请问是否与`C:\Users\Administrator\AppData\Local\Arduino15\packages\CH55xDuino\hardware\mcs51\0.0.14\cores\ch55xduino\USBCDC.c`中延时有关系，我添加了一个 delayMicroseconds(50000);有所改善。
`void setControlLineStateHandler(){
    controlLineState = Ep0Buffer[2];

    // We check DTR state to determine if host port is open (bit 0 of lineState).
    if ( ((controlLineState & 0x01) == 0) && (*((__xdata uint32_t *)LineCoding) == 1200) ){ //both linecoding and sdcc are little-endian

#if BOOT_LOAD_ADDR == 0x3800
        USB_CTRL = 0;
        EA = 0;                     //Disabling all interrupts is required.
        TMOD = 0;
        delayMicroseconds(50000);
        delayMicroseconds(50000);
       delayMicroseconds(50000);

        __asm__ ("lcall #0x3800");  //Jump to bootloader code
        
        while(1);
#elif defined(CH559) && (BOOT_LOAD_ADDR == 0xF400)
        USB_CTRL = 0;
        EA = 0;                     //Disabling all interrupts is required.
        delayMicroseconds(50000);
        delayMicroseconds(50000);
        delayMicroseconds(50000);

        __asm__ ("lcall #0xF400");  //Jump to bootloader code
        
        while(1);
#elif BOOT_LOAD_ADDR == 0xF400
        //todo: not working well, CH549 doesn't support direct jump
#endif
        
    }
    
}
`

DeqingSun, 2022-10-07:
@perseverance51 
Thanks for reporting the issue.
That is very likely to be caused by the erase command in https://github.com/DeqingSun/vnproch551/blob/master/main.cpp
The current version only erase 8 pages (8*1024) for CH559. And you get error on pack 146 (146*56=8176) because it was not erased before flashing.

I'll take a look and fix it tonight.

DeqingSun, 2022-10-08:
@perseverance51 
The new 0.0.15 release fixes the flash issue on CH559.

closed 2022-12-02 by DeqingSun