DeqingSun/ch55xduino #92: Some questions about program upload

ei4869, 2022-10-26:
First,I appreciate your great contribute to ch5xx.Recently I encountered some questions on my keyboard project.The first question was about upload my program,the upload is successful only when I use a blank chip.But when I tried again,it prompted that
""Load file as hex
   Loaded 4744 bytes between: 0000 to 12D1
ch375Version 35
CH375GetUsbID 0
DeviceVersion of CH55x: 2.50 
Send Detect: Fail
上传项目出错"
I made sure the libusb was installed and it stayed high between vcc and 5v while uploading.

The second question was the button didn't work when the ch5xxduino had connected to my cmputer and its driver was hibusb.But some of the program about led blink did work when I long press the button.some of my program:
void KeyModeSelect(byte mode) {
  if (Stop1 == 0 && Stop2 == 0) {
    if (mode == 0) {
      switch (Button_State(BUTTON1_PIN, &button1PressPrev)) {
        case Click:
        case DualClick:
        case LongClick: Keyboard_write(0xB0); break;
      }
      switch (Button_State(BUTTON2_PIN, &button2PressPrev)) {
        case Click:
        case DualClick:
        case LongClick: Keyboard_write('w'); break;
      }
      switch (Button_State(BUTTON3_PIN, &button3PressPrev)) {
        case Click:
        case DualClick:
        case LongClick: Keyboard_write(0xB1); break;
      }
      switch (Button_State(BUTTON4_PIN, &button4PressPrev)) {
        case Click:
        case DualClick:
        case LongClick: Keyboard_write('d'); break;
      }
      switch (Button_State(BUTTON5_PIN, &button5PressPrev)) {
        case Click:
        case DualClick:
        case LongClick: Keyboard_write('s'); break;
      }
      switch (Button_State(BUTTON6_PIN, &button6PressPrev)) {
        case Click:
        case DualClick:
        case LongClick: Keyboard_write('a'); break;
      }
    }
And I press my button but no response in computer.
I wish someone can help me,thanks!

DeqingSun, 2022-10-26:
Please upload the device manager screenshot when you upload, and screenshot of your Arduino settings.

Then you format your code in the question.

ei4869, 2022-10-27:
Thanks for your reply!The second question I have solved,I doubted that's because I put the hidkeybordmouse.h and hidkeyboard.h in the same folder.I reupload my code and it worked well.
The first question I still haven't solved it.But I use the newest official software wchisptool and it uploaded successfully.
My arduino settings:
[image]

DeqingSun, 2022-10-27:
@ei4869 

Make sure you send the chip to bootloader mode manually because it will not enter bootloader automatically in "USB setting: USER CODE". And you need to upload within 10 seconds or a V2.5 version bootloader will timeout.  

You mentioned that you installed libusb, it is not necessary. The problem may be caused by driver conflict. 

Try following: 

Connect CH552 to computer with booloader mode, open "device manager" and uninstall the device AND driver. You may need to replug and do it multiple times until it is automatically load the driver from WCH.

Here is a screenshot of my upload.

[image]

ei4869, 2022-10-31:
@DeqingSun 
You're right,I followed your instructions,and it worked well.The lisusb drivers is really not necessary.I think the most important thing to pay attention to is exactly what you said about uploading within 10s.
Anyway,thanks for you instructions!

ei4869, 2022-10-31:
Another questions,can I only use the hidkeyboardmouse  instead of hidkeyboard in order to use the mouse operation and keyboard operation in the same program?

DeqingSun, 2022-10-31:
@ei4869 The examples do not have dependency on another example. So you can use HidKeyboardMouse to use both mouse and keyboard.

ei4869, 2022-10-31:
@DeqingSun 
That's great!Thanks for your awesome work again!

closed 2022-10-31 by ei4869