DeqingSun/ch55xduino #95: Timer 1 overflow vector not working

StinosTheEngineer, 2022-11-29:
Im trying to get timer 1 to work. Until now it is not entering the overflow interupt. My clock settings are:
//Clock settings
  TMOD |= (1 << bT1_M1);
  TMOD &= ~(1 << bT1_M0); //timer 1 in 8bit mode

  TH1 = 251; //T0_CYCLE = 5 --> TH1 = 256-T0_CYCLE;
  TCON |= (1 << TR1) | (1 << IE1);
  IE |= (1 << ET1) | (1 << EA);
  TMOD &= ~( (1 << bT1_GATE) | (1 << bT1_CT) ); // &= ~(1 << bT1_GATE) & ~(1 << bT1_CT);
  T2MOD &= ~(1 << bT1_CLK);

Then I tried to get the interupt to work inside wiring.c:
void Timer1Interrupt(void) __interrupt(INT_NO_TMR1) __using(3) {
 //some code   
}

I also tried some other stuff but thats too much to talk about. My question is what am I doing wrong. How do I enter an overflow interupt on timer 1.

DeqingSun, 2022-11-29:
Timer1 is generally used for Uart0 timing. But if you really want to use it, put the interrupt handler or at least a declaration of function in main.c . That is the limitation of SDCC compiler.

StinosTheEngineer, 2022-12-05:
Yes that is what I tried but that didn't work either. I tried to copy something that was similar of the prototype of timer 0: void Timer1Interrupt(void) __interrupt(INT_NO_TMR1) __using(3);

DeqingSun, 2022-12-05:
If you can not get timer2 working, here is example showing using timer2.
https://github.com/DeqingSun/ch55xduino/blob/ch55xduino/ch55xduino/ch55x/libraries/Generic_Examples/examples/05.USB/USBtinySPI_CH552/USBtinySPI_CH552.ino

Please check your register settings.

closed 2022-12-05 by DeqingSun

StinosTheEngineer, 2022-12-05:
I don't see any timer2 registers edited in this file. In my file the registers are setup like follows:
 IE |= (1 << EA); //enable interrupts

  T2CON |= (1 << TR2); 
  T2CON &= ~(1 << RCLK) & ~(1 << TCLK) & ~(1 << C_T2) & ~(1 << CP_RL2); 
  T2MOD &= ~(1 << bT2_CLK);

StinosTheEngineer, 2022-12-05:
It works now! Thank you for the help. Sadly, I bricked the MCU somehow. The COM port disappeared.

DeqingSun, 2022-12-05:
It is very likely that your code interfered the USB transaction. Pull up the D+ and plug in your board to force it into bootloader mode, if you are using development board there should be a button to do the pull up.

StinosTheEngineer, 2022-12-06:
I have a tough time figuring out what the mistake is. Pulling up D+ doesn't do the job. Does the USB transaction run on UART0? Still if that is the case it shouldn't brick the device as the Baud rate generator is on Timer1 by default. It all seems to happen when ET2 is enabled.

DeqingSun, 2022-12-06:
Can you find the bootloader device in your device manager on your computer when you enter the bootloader?

DeqingSun, 2022-12-06:
Did you clear overflow flag by code? It won't clear automatically and will repeat interrupt if you don't clear.

StinosTheEngineer, 2022-12-06:
That was the problem indeed. It works perfectly now! Thank you for your help.