DeqingSun/ch55xduino #99: IDE doesn't want to compile

StinosTheEngineer, 2022-12-20:
I installed the package for the arduino IDE but when I try to compile it says the following:

Preproc file not found: No error
cp:: No such file or directory
Preproc file not found: No error
cp:: No such file or directory
Preproc file not found: No error
cp:: No such file or directory
exit status 1
Fout bij het compileren voor board CH552 Board

StinosTheEngineer, 2022-12-27:
I fixed this isue. Something with antivirus software. I face a new problem now: The moment I want to upload any code the MCU shuts down. After a few seconds it restarts but the uploading failed by then. Booting the MCU into the bootloader doesnt work

StinosTheEngineer, 2022-12-27:
This is the error message: 

CH55x Programmer by Deqing
Updated on: 2022/10/08
------------------------------------------------------------------
Load file as hex
   Loaded 4074 bytes between: 0000 to 1034
ch375Version 35
CH375GetUsbID 0
No CH55x USB Found, retry...
No CH55x USB Found, retry...
No CH55x USB Found, retry...
No CH55x USB Found, retry...
No CH55x USB Found, retry...
No CH55x USB Found, retry...
No CH55x USB Found, retry...
No CH55x USB Found, retry...
No CH55x USB Found, retry...
No CH55x USB Found, retry...
No CH55x USB Found, retry...
No CH55x USB Found, retry...
No CH55x USB Found, retry...
No CH55x USB Found, retry...
No CH55x USB Found, retry...
No CH55x USB Found, retry...
No CH55x USB Found, retry...
No CH55x USB Found, retry...
No CH55x USB Found, retry...
No CH55x USB Found, retry...
Found no CH55x USB
Time limit reached, exit process

XanCraft21, 2022-12-30:
I had a similar problem like in the first comment here. The file cp was not found. I looked through the files in the folder %appdata% (can be only found on windows by searching in app search) and I couldn’t find anything. Then all of a sudden the error was gone, and it compiled fine. I am not sure what happened. I haven’t tried uploading yet because i got frustrated. I will wait until i get a couple extra boards for testing.

dwillmore, 2023-02-15:
The Readme.md says:

> If your board is never used with ch55xduino before, you need to make the ch55x chip enter bootloader mode. You need > to disconnect USB and unpower ch55x, connect the pull-up resistor on D+ line (generally a 10K resistor between D+ and > 5V, controlled by a push-button or adjacent pads). Then you connect USB. and hit Upload. Also, a blank new chip will enter the bootloader automatically.
> If you have used ch55xduino once and your code doesn't crash the USB subsystem, you can simply press Upload. Arduino and the firmware will kick the chip into the bootloader automatically.

I had the results you show on my board.  I followed the directions by unpowering my board, pressing the download key which ties D+/P3.6 to +V with a 10K resistor.  While holding down the button, I pressued upload in the IDE and the schetch successfully uploaded.  

Unlike the second part of the instructions, it seems I have to do this every time--not just the first time I use it with ch55duino.  So, I may still be doing something wrong.