DeqingSun/ch55xduino #103: Mass storage: support for files larger than 32kB

buttim, 2023-01-02:
Is it possible to support files bigger than 32kB though read callbacks?

DeqingSun, 2023-01-02:
Not right now. You would need to improve USBMediumAccess.c .
The 32kB limit is to make the address->file calculation easier.

buttim, 2023-01-02:
Thanks for your quick reply. So if I understand it is not a somehow implicit limit.
I will give it a try.
Any hint is welcome obviously ;)

DeqingSun, 2023-01-02:
I don't quite remember how the code work since it has been a long time. But it seems each file is reserved for 64 512-byte clusters so that leads to the 32K limits. 

If you want to use a larger file, you would need:

1. change start cluster of files in ```LUN_Read_func_Root_DIR```

```uint16_t startClusters = 2 + fileIndex*64;  //make file 64 clusters, 32K apart. Should be enough.```

2. change any functions refers to filesize of filesOnDrive. Make the FAT table correct and the data access point to the correct cluster.

I can think of 2 ways to do improvement, the easier way is to make files further apart, such as 128 or 256 clusters per file, you may simply change the "64" and "32768" in the files when you understand how the code works. 

Or if you want even larger files that mapping and FAT-generating code would need to be more complicated to lay the clusters more efficiently.