DeqingSun/ch55xduino #108: How to read the serial port data and add it to the array？

15010200730, 2023-01-24:
Happy Spring Festival.
How to read the serial port data and add it to the array？I want to add all the received data to the array，But it diden't work,This my progrems,If can, please help me modify it, thank you！

```cpp
/*
  UsbSerialAdaptor

  A simple example turns CH552 to a serial adaptor, 
  relaying data between USB serial and Serial0.

  Baudrate of Serial0 is determined by USB serial's rate.

  created 2020
  by Deqing Sun for use with CH55xduino

  This example code is in the public domain.

*/

extern __xdata uint8_t LineCoding[]; //lineCoding of CDC is located in this array
__xdata uint32_t oldBaudRate = 9600;

char test[5];//存放用的数组

void setup() {
  Serial1_begin(9600);
}

void loop() {

int a = Serial1_available();
if(a>0){
  for(int num=0;num<a;num++);
  {
    test[i] = Serial1_read();  //这里读Serial1_available的变量a的长度是1，无法将所有接收到的数据依次添加到数组
  }
  __xdata uint32_t currentBaudRate = *((__xdata uint32_t *)LineCoding); //both linecoding and sdcc are little-endian
}

}

```

DeqingSun, 2023-01-24:
This is not a C learning playground.

closed 2023-01-24 by DeqingSun

15010200730, 2023-01-25:
> This is not a C learning playground.

This have a bug。。。please check Serial1_read() function and Serial1_available() Whether it is normal.。。