DeqingSun/ch55xduino #109: Low Power to 0.15mA?

sabas1080, 2023-01-25:
Hi

I am trying to achieve sleep mode at the lowest consumption, but I have only achieved 6mA, the datasheet mentions 0.15mA or 0.7mA

[image]

My configuration is:

```
pinMode(11, INPUT );
  for (uint32_t i = 14 ; i < 18 ; i++) {
    pinMode(i, INPUT );
    //digitalWrite(i, LOW);
  }
  pinMode(30, INPUT );
  pinMode(31, INPUT );
  for (uint32_t i = 34 ; i < 38 ; i++) {
    pinMode(i, INPUT );
    //digitalWrite(i, LOW);
  }
  pinMode(ledGreen, OUTPUT);
  pinMode(ledRed, OUTPUT);
  pinMode(ledYellow, OUTPUT);
  pinMode(interruptPin1, INPUT_PULLUP);
  pinMode(interruptPin2, INPUT_PULLUP);

  SAFE_MOD = 0x55;
  SAFE_MOD = 0xaa;
  GLOBAL_CFG = GLOBAL_CFG | bLDO3V3_OFF;//3.3V LDO OFF 
  WAKE_CTRL = WAKE_CTRL |bWAK_RST_HI ;
  SAFE_MOD = 0x00;

after mode Sleep
```
USB Settings is USER CODE w/ 0b ram

Thanks

DeqingSun, 2023-01-26:
I do not have a setup to test the current for now. But I can help you to do some check and see if I can find some tutorial for you. [this link](https://www.wch.cn/bbs/thread-90171-1.html) did not mention much about the software setup so I think there are some thing you can check.

Do you have LED and it is ON?

Do you have on board external LDO? Did you measure the current before it?

Can you share the schematic and a photo of your circuit?

sabas1080, 2023-01-26:
Do you have LED and it is ON? No
Do you have on board external LDO? No 

Can you share the schematic and a photo of your circuit?
Schematic
[image]

DeqingSun, 2023-01-26:
The schematic seems OK if you measure current on the battery side.

If you get 6mA, it looks like the working current of 12Mhz@3.3V. Would you try to see if you have a difference in current between doing/not doing sleep?

sabas1080, 2023-01-26:
i have made it, 10uA!! Thanks @DeqingSun

closed 2023-01-26 by sabas1080

kahoch, 2023-02-01:
> i have made it, 10uA!! Thanks @DeqingSun

@sabas1080 Amazing! How do you achieve that?

sabas1080, 2023-02-05:
@kahoch with example schematic

> Do you have LED and it is ON? No Do you have on board external LDO? No
> 
> Can you share the schematic and a photo of your circuit? Schematic [image]

and example code https://github.com/DeqingSun/ch55xduino/issues/106