DeqingSun/ch55xduino #112: How to use USBCDC and HID together?

yuan910715, 2023-01-30:
Hi
I want to use CH552 to finish a 6 keys keyboard , I write the code reference to the example HidKeyboard , and it works fine.
I want to develop a PC software , it can communication to CH552, so I can define the keyboard button to any key ,no need to upload the program again to change the key,
I use USBCDC to finish serial communication , but HidKeyboard is not work
How to use USBCDC and HID together?
Or is there any other way?
Thanks a lot

DeqingSun, 2023-01-30:
It is not a good practice to use CDC just for reconfigure the keyboard map. Adding HID raw report should be a better choice.

Take a look at https://github.com/yswallow/CH552duinoKeyboard

If you still want to use CDC and HID together, you may check 

https://github.com/Noteolvides/CubeTouch/tree/main/Examples/CDC_KEYBOARD

I've not tested any of the referenced project and I don't know how they work or if they were buggy.

yuan910715, 2023-01-30:
> 

I tried https://github.com/Noteolvides/CubeTouch/tree/main/Examples/CDC_KEYBOARD 
it works well, next I will develop a PC program to reconfigure the keyboard map , so I can give the 6-key keyboard as a gift to my friend , and they can easily reconfigure the keyboard map
Thank you!!

I'm a little confused: what do you mean 'Adding HID raw report'?Is that modify the arduino code when need reconfigure the keyboard map?

DeqingSun, 2023-01-30:
The same hid endpoint can do keyboard, mouse and raw hid communication with a different report id. You would be able to use hid API on computer to do 2 way communication. That is how Marco pad keyboard do remap.

yuan910715, 2023-02-01:
I have completed a small three-key keyboard, and can reconfigure the keyboard map through the serial port.
Now waiting for the production of PCB , I will show the pictures and open source in a few days， please wait..
Thanks!!

yuan910715, 2023-02-02:
Hi, I used https://github.com/Noteolvides/CubeTouch/blob/main/Examples/CDC_KEYBOARD/CDC_KEYBOARD.ino as example.
I see the Ep3Buffer is outside the reserved space.
But why I compiled two .hex file ,one is Tools--> USB Settings--> USER CODE w/148 USB Ram and the other is USER CODE w/266 USB Ram , can both works well ,I don't know why.

DeqingSun, 2023-02-02:
The memory overlap may not have immediate problem, bugs will be triggered in some boundary conditions.

Using 266 will be safer and avoid the problem.

yuan910715, 2023-02-07:
hi , I completed my DIY project , https://github.com/yuan910715/3keys
Thank you very much, I can't do it without your help.

closed 2023-02-25 by DeqingSun